Ralim/IronOS #2: Suggestions for improved button handling

jsobell, 2017-05-14:
This system is so much better than the factory code that I installed it on the first day. Perhaps that's the geek in me, but then that's why I bought the TS100!

Re the button detection and usage...
I can see that using a timer based detection solves debounce issues nicely, but it makes changing values a bit hit-and-miss. When you try and change any value by a single increment it's a case of trying to judge the perfect time press, as too short and you have to start again, too long and you do the same dance in the down direction.
Given that at almost any point the button features are performing a single task, would it be feasible to add a flag to indicate "new button press" that is cleared by any code handling that press? This would allow repeated presses to be handled instantly, bypassing the need to hold the button to go up 50°.
Detection of the button being released can also be used to reset the initial state, so the original time detection can be retained if you feel it's needed.
I guess if detection of double-presses is an issue then the button pressed flag could be triggered by the release, rather than the press.

Just a suggestion, as the button press stuff is the only minor niggle I had.

Ralim, 2017-05-15:
Hi,
I had wondered if this was an issue (After using the base software and this fork for a while, I guess I have found the right timing).

I will look into a better handling for the button presses, I did originally want to move them to be interrupt and clear based (similar to your suggestion). 

I'll update when I have a chance to look at implementing this :)

Ralim, 2017-05-15:
Hi,
I have pushed up a pre-release firmware that reduces the time out on button presses.
Could you let me know if this is better or worse from your point of view?

Ralim, 2017-05-17:
As far as I know the latest release should improve buttons dramatically.
Closing this issue, for now, feel free to open it again if you want anything changed.

closed 2017-05-17 by Ralim

jsobell, 2017-05-28:
Finally got around to trying the latest release. That is excellent, and a much nicer and responsive input process.
I like the standby graphics, and the sensitivity option is an excellent idea, as my iron comes out of standby if I type on my keyboard on the same desk.  Does this settings also apply to the screen-saver?
Up/down arrows look good.

Ralim, 2017-05-28:
Glad to hear its better, It was actually fairly simple. 
I will look at adding auto-repeat in later on, but for now, this is nicer to use.
The sensitivity affects everything because of the current setup.
The accelerometer has a built in function for detecting movement, so the sensitivity just sets the threshold for the built-in function. Let me know if you think these levels should be changed, as they are completely open to user testing at the moment. (I chose the values fairly arbitrarily)