Ralim/IronOS #3: Improve soldering info screen

Ralim, 2017-05-18:
Need to make it look better, and possibly display input voltage or similar as well.

closed 2017-05-21 by Ralim

CptZeroG, 2017-07-19:
Hi Ralim, 
I don't understand: have you implemented the display of input voltage? It would be very usefull :)

Ralim, 2017-07-20:
Yes, 
It's under the Extras menu.
You can access it via the following steps:

1. On the idle screen press both buttons
2. The unit will now show the current tip temperature
3. Press the button near the tip of the soldering Iron
4. The unit will now show the input voltage that the Iron is measuring

For further information, there are some details in the readme on the main page of this repository.

CptZeroG, 2017-07-21:
Ah ok, thanks Ralim, i did't found on the menu schema.

P.s. I hope that you add the custom logo feature...it's a great Fw :) complimets

commit cb5d4bc4 (2017-05-21): Import up/down indicator from miniware

Closes #3 with nicer heating display