Ralim/IronOS #6: Left hand display does not work.

iLLiac4, 2017-05-21:
Hi. In your latest release flpdsp has no effect. So settings to flip display to left/right hand is not working it is always for right one.

Ralim, 2017-05-21:
Flip display is working well for me. 
Hmm.
Can you confirm, that you have taken the following steps:

1. Enter the settings menu, cycle through and change FLPDISP to T (True)
2. Continue through the settings to the end of the menu, and you are returned to the idle display
3. Power cycle the Iron (unplug and re-plug)
4. Display should now be flipped

If it works then close the issue, otherwise let me know and ill look into the issue further :)

iLLiac4, 2017-05-21:
Ok with those steps it works.

closed 2017-05-21 by iLLiac4