Ralim/IronOS #10: Possible to set the max temp to 450?

OtterNas3, 2017-06-13:
Just a short question if it would be possible to crank up the max temp to 450, only for a short period maybe 1 minute or 2 then fall back to 400 to not burn out the heater?

Appreciate your work here ;)

Ralim, 2017-06-13:
It should be possible technically. I just have no idea how the tip will handle the higher temperatures.
How would you propose activating such a feature? I'm more than happy to look at creating a test build so that you can test the higher temperatures on your iron, I just can't see and easy way to integrate this into the control scheme of the iron when you are soldering?

Thank you :)

Maartenwut, 2017-06-13:
I run mine on 450 a lot of the time, works without a problem! (Though the body will get a little warm after a while)

Ralim, 2017-06-13:
Awesome :)
I'm glad to hear the 450 degrees doesn't cause too much of an issue.
Any idea how best to have the boost mode added to the software?

OtterNas3, 2017-06-14:
Hey Guys,

maybe it is possible to make a Double-Tap on the button next to the tip in Soldering mode, then it cranks up to 450 for set time (maybe even configurable in the settings menu)?

Ralim, 2017-06-14:
Hmm sounds like a possible way to handle it.
Unfortunately, I'm fairly tied up until end of next week so may not be able to add it until then :(
More than open to any other suggestions on how to handle this, though double tap probably makes the most sense. Would another option be to have it detect holding the front button and have it boost the temp while you hold the button?

Or we could even go for just having it drive the tip at maximum while you hold the button for when you have one of those big solder jobs?

OtterNas3, 2017-06-14:
Isn't the Hold-Down of that button alrdy a function?

Ralim, 2017-06-14:
Not currently, pressing either button takes you to the temperature screen, and pressing both at once exits soldering. I could add a boost mode to long holding the button.

To compile, I use System Workbench from openstm32.org. You should be able to import the workspace folder into it. However, I do plan to document this better on the wiki... Though if you do get the code compiling I would appreciate anyone who would like to write it up :)

OtterNas3, 2017-06-14:
I'll give it a try!

Bucknalla, 2017-06-30:
Just seen this - I'm also happy to help with a write-up!

Ralim, 2017-07-07:
Hi, @OtterNas3 
I've been thinking on this, would it be acceptable to add an option to just remap the button near the tip to a temporary 450C boost mode? This would make it so that only the other button could enter the temperature settings screen. I could build a firmware fork for testing this option if that could work.
Mostly just because I haven't had the time to re-code the code for button handling yet.

OtterNas3, 2017-07-07:
@Ralim 
That would be totally fine, as you normally dont change the temp that often.
I would make it like a Hold-For-Boost button, and if possible Double-Tap for Temp setting.
Tried alrdy to implement but, i am not rly a good coder, more a Hardware guy :)

Thanks for getting into it!

Ralim, 2017-07-07:
I'll make it a hold for boost (Its easiest) 😀 
Yeah, no problem, I'm not amazing, but I get there eventually :)

closed 2017-07-08 by Ralim

Ralim, 2017-07-08:
Let me know if you have any further ideas for this (aka feel free to re-open the issue if you have more ideas)

OtterNas3, 2017-07-08:
Tested and it works beautyful!

commit d536fb33 (2017-07-08): V1.11 - Adding a Boost mode

Allows users to enable a boost mode.
This allows you to hold down the button near the tip while soldering to
temporarily boost the soldering temperature.
Closes #10