Ralim/IronOS #12: Compiling Firmware for STM32

Bucknalla, 2017-06-30:
Hi!

I'm new to developing for the STM32 and am interested in adding some features to this project. What tools do you recommend for compiling/developing with your library? 

Thanks!

Ralim, 2017-06-30:
Currently, the project is compiled using [System Workbench for STM32](openstm32.org).
To get it compiling, simply fork the repository to your account, then clone to your computer. Then in System Workbench set the workspace folder to be the workspace folder in this repository. Then once you open System Workbench you can then Import existing projects into workspace and find this project.

The build files are all set up to produce a hex file that can be programmed onto the Iron using the bootloader.

Bucknalla, 2017-06-30:
Brilliant! Thanks!

closed 2017-06-30 by Bucknalla