Ralim/IronOS #13: Slow down temperature reading

Careless, 2017-07-03:
Is it difficult to perhaps implement a setting that allows the update/refresh rate of the temperature reading on the main-screen to be slower to update?

Perhaps a High / Medium / Low setting? My father liked the way the iron was with its stock firmware, but now he thinks its broken because even when it hits 400 or 450 it's constantly fluctuating rapidly. 

In fact, I would suggest instead "TEMPRND" for "Temperature Rounding" True or False. When TRUE, temps would be rounded to the nearest 10*C.

I don't mind the jumpy screen, but I do agree with him that it's a little "epileptic" in its update rate.

Ralim, 2017-07-04:
I think slowing down the screen update rate would make sense. 
This should not be too hard to compliment, and I can look into this when I have time :)

Temperature rounding could be an interesting idea to add to the firmware, I'll think on that.

The iron is meant to remember the last used temperature normally. (When you exit the set temperature screen it saves the chosen temperature to flash). Is that what your referring to ?

Careless, 2017-07-04:
I used the iron yesterday despite the temperature reading fluctuations and it went to 450 degrees or so, because that what I had it set to the previous time. So I see what you are referring to as per the temperature memory on restart.

Though the original firmware had a config text file that allowed the user to set the default temperature which I thought was nice.

Either way, I understand how that works now so it's not as much of an issue. I think the current implementation with the memory of the previous setting is fine.

Ralim, 2017-07-07:
Hi,
I've added a slowed down display mode in the newest release.
Do you think it would be worth adding a medium speed as well ?

Careless, 2017-07-07:
Hi!

I just gave it a test run. It is far better than it was before.

May I may another suggestion to improve it further?

- TMPUNIT -> Temperature Unit, C or F
- TMPDISP -> Slow, Medium, Fast
- TMPRND -> 1, 5, 10

I only ask this because after testing all the modes, I think medium speed (between current slow and fast) would be the best, and Rounded mode is like Fast mode but with large jumps in between.

If I were to choose only one setting, It would be Medium, Rounded to 5 degrees.

Otherwise, excellent implementation.

Is there somewhere I can donate a slice of pizza, cup of coffee/beer/tea whatever?

Ralim, 2017-07-07:
Hi,
Yeah, I think I agree with your idea after playing with it a bit more.
I'll add that in shortly (probably 10-20 mins, maybe sooner) after I finish the current addition.

Nowhere publicly to donate at the moment, though you can use this if it's easy for you:
https://paypal.me/RalimTek
I'll probably add something to the readme eventually, though I started this mostly because I liked the hardware 😃

kokein, 2017-07-07:
Keep Up good work !! :)
I've also made a donation. cheers !!
Nick

Ralim, 2017-07-07:
Thank you 😁 
I have pushed a new version 1.10 for you to try out and look at.
Let me know what you think of the options now, more than happy to keep tweaking :)

Careless, 2017-07-07:
Version 1.10 is a winner!

I personally like the Slow Update Speed, and the Rounding to 10 degrees after trying it out. Seems accurate enough and slow enough to not feel like an epileptic seizure.

The setting descriptions are a nice touch. Perhaps putting the units in which the settings are shown in the description would be good. like SLPTME could say "SLEEP TIME (MINUTES)".

I'm sure it will be easy enough for most of the user to figure out, but it would be a nice little detail for a future release.

So glad that updating the firmware takes like 2 seconds too.

Ralim, 2017-07-07:
Awesome!
I'm one of the weird people who like it to be really fast, so thank you for bringing this to my attention in the first place.

Good idea, I'll add in the units in the next version definitely.

I think if upgrading wasn't this fast most of this wouldn't have happened :P

I'll leave the issue open for a while, but close it when you're happy with it. Feel free to suggest any refinements to the speeds if you wish though :)

Careless, 2017-07-07:
The only thing I would change regarding the temperature display is the sensitivity, if it's even possible. for example, sometimes it shows 340*C and then drops to 280*C for one update cycle, and then goes up and down and up and down with a large variance.

I don't know if that's something you can tweak, so I didn't mention it. It would improve the readout accuracy, but I don't mind how it is now.

Just a little jumpy.

Ralim, 2017-07-07:
I've just pushed a small release update that improves the pre-PID filtering. 
Could you give this a try to see how it improves the jumpiness for you?

I haven't seen it oscillating that large of an amount. How are you powering the Iron?
I could also have it force a screen update if the temperature swing is too large to avoid it looking as wildly swinging?

Careless, 2017-07-07:
I'm using a 18V and a 21V DC power supply. I'll test the new update shortly.

Careless, 2017-07-07:
I just tested with Slow update speed and 1 Degree rounding, and it works amazing. Even better than 5 or 10 degree rounding on any other speed.

In fact, I'd go so far as to say it would be almost worthwhile to remove all the options and just this setting and not make it changeable because it works so well now, but you've done all the work now.

But this is accurate, perfect speed, and will be what I use from now no.

Thanks again!

Ralim, 2017-07-07:
Awesome!
That's the best news 😆 
I'll leave all the settings there for now, since I've added them, and some people might still like them.
I had been meaning to update the tuning for ages, and completely forgot earlier that it would probably help with this. Glad to hear its better though.

I'll close off this issue for now, though open it again if you ever have any issues :)

closed 2017-07-07 by Ralim