Ralim/IronOS #14: Logo :)

kokein, 2017-07-03:
Hi :) 
Any way to add logo option like in oryginal firmware? 
Oryginal one want "fit", my pc says that there is not enough space.

Ralim, 2017-07-04:
Hi,
Currently, the firmware does not support a custom logo without you re-compiling the firmware yourself. As the implementation for the virtual USB flash drive does not fit on the microcontroller when using the open source toolchains.

Eventually, I plan to add a way to use an extra software tool to edit a logo onto the Iron, I just have not had the time (yet) to get around to this.

Don't try to upload any files other than upgrade files to the DFU bootloader. (I cannot guarantee that this is safe)

kokein, 2017-07-04:
It's not necessary :)
But will be nice feature.
Thanks! BTW it's VERY nice firmware!

Ralim, 2017-07-12:
Just as an update to this, I have a concept for a method of allowing you to set a custom bootup image via using a program to create a special .HEX file you drag into the bootloader after programming the firmware. I'll work on this more when I have more time to create the program. So no promise of this feature just yet, but its in the pipeline eventually.

kokein, 2017-07-12:
Ok :)
Till now it's SUPERB software ;)
Thanx!

M0LIN0, 2017-07-16:
Hey Guys
First of all, I'd like to thank you very much for your effort, you're investing in this great firmware, I absolutely love it, especially the Boost-Button!
I'm also interested in having a personal logo, as the original firmware allows me to do so and I think, I found an easy way to upload the logo-file.

Here is what I did:
First of all, I installed the latest original firmware APPV217.hex and uploaded my custom logo as LOGOIN.bmp via the virtual USB flash drive and it worked fine.
I then installed your firmware and of course, there was no logo displayed any more.
But when I re-installed the original firmware, I expected something interesting: The LOGOIN.bmp was still there! So I think the file was never deleted while your firmware was active.
Therefore I think you only have to get your firmware to search for an existing LOGOIN.bmp and display it!
Maybe this helps, I thought...
Best regards

Ralim, 2017-07-17:
Hi,
You have actually found the basic idea behind my plan to allow for custom logos on the iron.

To explain why this behaviour occurs, the original firmware uses the last ~30K for the virtual USB drive.
When you program a new firmware using the DFU mode, it only replaces the parts that have changed. As this firmware is less than the ~30K before the virtual USB it does not overwrite the existing files.

My plan I am currently working on is creating a small app to convert your source image to a .HEX file that can be programmed using DFU. This will always place the logo in the same location, which will make it easy to code. 

This is a little bit annoying to use another program, and it will probably be windows only at first :( However, it is the easiest method for me to get up and running. Also once you make the .HEX of your logo, you can use it in the future without any issue.

Eventually, I would like to support an easier means to setup the logo, however, whenever I try and get a virtual USB working with a FAT32 filesystem I run out of program space, as the open source GCC compiler just doesn't optimise quite as well as the proprietary IAR one.

Currently been a bit busy, but I do plan to support custom logos soon :)

mkninc, 2017-07-17:
Regarding the resulting code size, have you tried „-Os“ as  optimisation? As I have seen your are using „-O3“ in the Release configuration. 
You could also try „-flto“.

Ralim, 2017-07-23:
@kokein @M0LIN0 @Adrianotiger @mkninc 
Hi,
Letting you all know that the latest release is the beginning of the support for custom logos.
I have pushed up V1.15 which adds support for loading a logo if it has been programmed into the unit.
There is also a C# app that makes the custom .HEX file for loading the logo onto the unit.

There are instructions on the main page for how to make the .HEX file you need to load onto the unit while in DFU mode.

*Please* let me know what issues you have using this, I don't have a massive set of pictures to test with, nor do I have hours to make and test them at the moment.
If you don't have a way to make the HEX file from your image (ie if you can't run the program), for the first little bit ill be happy to make them for you to test with. Though I won't be able to offer this forever.

@mkninc 
It makes little to no difference (few bytes). The open source compilers for arm M series micro controllers is just not that amazing compared to the paid ones. But it works excellently for the rest of the system.
Performance is slightly better on the -O2 or -O3 than -Os. But it's nearly no difference at all, I *have* tried them before in the past, and every so often I try them again to test any performance change.

kokein, 2017-07-23:
Ok. Thanx! Will try later =)

M0LIN0, 2017-07-23:
Hey Ralim
Thank you so much! I just flashed the latest firmware and tested the logo editor application and it worked perfectly at the first try!
I think the logo editor is easy to use and well arranged. I also had to use the "invert" fuction, which worked perfectly.
Thank you very much for the effort, you're putting into this firmware! Is there any possibility, we can support you with a small donation as for example paetron?
Best regards
MOLINO

Ralim, 2017-07-23:
Glad to hear it worked!
Eventually, I'll make the threshold for what it calls white or black adjustable but for now, it seems to work on all my test images. So very glad to hear it worked for someone else too :)

I don't really want to push a patreon at the moment, mostly because I don't want people to have the expectation that I'll add features regularly (I've run out of most of the ones I can think of). 
That said I did add a little note at the bottom of the readme that I do accept donations on https://paypal.me/RalimTek If you would, or if you would like another system (if you don't like PayPal etc), feel free the email me at any time.

Hopefully the firmware's working well for you.

kokein, 2017-07-24:
Works great ! :)
Thank You !
Nick / kokein

Ralim, 2017-07-24:
No problem :)
Glad to hear it worked.
I'm going to close this for now, but open it (anyone) if you find any more issues on the logo support :)

closed 2017-07-24 by Ralim

Ralim, 2017-10-27:
Hi,
Sorry about the delay getting back to you. 
Can you confirm that you were running 1.17.1 and could you also provide the hex file you uploaded to the iron so I can test locally?

Bluebrain2000, 2017-10-27:
I just tried it again to send you the logo I used, and now, suddenly it worked.
I did exactly the same like the last time.
Photoshop > png > converter > save .hex

Thanks anyway.

Ralim, 2017-10-28:
Normally this sort of temperamental behaviour is the DFU bootloader sadly :/
I have found under Linux that just attempting the flash multiple times works usually on the 2nd or 3rd time.

sibusaman, 2017-10-28:
In my experience, flashing fails the first time, always. Then the filesystem mounts again with *.ERR file, copying the *.hex file again into this resulted in success every time, until now.  

I haven't faced it yet, but flashing large hex files by this method could pose a problem.

Bluebrain2000, 2017-10-29:
I'll be back at my office PC on monday and check again with the logo file that didn't work.
But I definitely never got an .ERR file.
I flashed the firmware (which worked the first time) and then tried the logo (different) 2-3 times and I always got an .RDY file back, even though, no logo was displayed.

Ralim, 2017-10-29:
@Bluebrain2000 It would be good to have a reference of the logo files (both working and not) to look at for differences if you could :)

kokein, 2020-01-27:
Hey,

I just signed the petition "Ratujmy koale! Żądamy uznania ich za gatunek
zagrożony wyginięciem" and wanted to see if you could help by adding your
name.

Our goal is to reach 1,000,000 signatures and we need more support. You can
read more and sign the petition here:

http://chng.it/ZfvNM7T8Kh

Thanks!
Mikołaj