Ralim/IronOS #16: One more when click when the settings help opens up to close it again

twikis, 2017-07-07:
Hi!

Thank you for this aaawesome firmware for this super cool iron! Just great, so excited to see even more features in the future!

Here's one little issue I found: Could you please add a second click when the help shows up, so that you don't have to a) change the value the help shows up for or b) jump to the next settings item when you didn't remember the value? It is pretty easy to get used to remembering the value that it was set to before the help shows up, but I just think it's slightly more convenient.

Cheers!

Ralim, 2017-07-08:
Ah, yep.
Completely missed that issue last night. I'll get on it soon. 😄

Ralim, 2017-07-08:
@twikis Just pinging you to let you know the newest release should fix this behaviour 😄

commit 0a7b6145 (2017-07-08): V1.10.2

Closes #16
Fixes pressing a button to exit the settings prompts
closed 2017-07-08 by Ralim