Ralim/IronOS #17: Goes to sleep while soldering...motion detection true.

stray77, 2017-07-09:
I've only used it a few times but it went to sleep in my hand a few times.  Motion detection was true and I was soldering a couple rows of pins (STM32F103 minimum dev board).  Perhaps add motion sensitivity levels? 0=off, 1=low, 2=med, 3=high?

Ralim, 2017-07-09:
Hi, 
There already is motion sensitivity levels, however I'm currently thinking of increasing the adjustment ranges. Can you test and make sure your on the default of high sensitivity?

stray77, 2017-07-09:
Yes, it was on high (default) as I had just updated the firmware and I double checked just now.

twikis, 2017-07-09:
Actually I'm having the same issue, it's going to standby a bit to much and also I have to shake it sometimes to wake it up from standy (sensitivity is on H). Some more sensitivity would be great :+1:

Ralim, 2017-07-09:
Ah okay, I'll push up some next settings in the next day or two :)

twikis, 2017-07-09:
My TS100 is the SainSmart Pro32. Maybe there are differences between the models e.g. different components?

Maartenwut, 2017-07-09:
No.

Ralim, 2017-07-10:
@twikis All the irons are exactly the same!

@stray77 @twikis Can you please test the newly uploaded firmware?

stray77, 2017-07-11:
Went to sleep on me at 6, trying 8 next... (again doing a row of pins with very small adjustments from pin to pin, you know 1.54mm)

Ralim, 2017-07-11:
Damn, you must be moving a very small amount when doing the fine soldering.
The chip supports 127 levels for the motion detection feature, so there might need to be some adjustment for the mapping from the 1-8 levels to the values. A level of 7 is the same as the original firmware's value if I remember correctly.

twikis, 2017-07-11:
Solved it for me so far, although I wouldn't mind a level 10 if possible... :)

Ralim, 2017-07-12:
At level 9 the accelerometer is at the most sensitive possible.
However, I think I have an idea of what might be causing this.

Ralim, 2017-07-12:
I have pushed up a beta test file for 1.13-alpha, could you please try this @twikis and see how it feels?

stray77, 2017-07-14:
I'm glad you found an issue with 1.12 because I found it as bad or worse than previous versions.  1.13 seems to be a big improvement as far as going to sleep in your hand,  I'll let you know if it goes to sleep on me again.  I did at one point manage to get a bunch of zero's and three's scrolling across the display with no way out except pulling the power.  I'm not sure how I cause it but thought I'd mention it in hopes you'd recognize the issue.  If I'm able to replicate it I'll open a new issue with specifics. Thanks again for your efforts.

Ralim, 2017-07-14:
I'm glad it is much improved!
Please let me know if the issue comes up again, its something I have to remember to watch carefully as I add other features.

That issue shows up when code tries to print something to the screen that isn't in the font. I haven't seen it personally for a while, so this is troubling. If you find any steps to get near the problem please let me know :)

Ralim, 2017-07-17:
@stray77 Hows the motion sensitivity feeling?

Ralim, 2017-08-03:
Closing this issue, as the sensitivity scale has been adjusted to cover the whole range of the IC. If the iron is still going to sleep on you at max sensitivity then there isn't much more we can really do, as it's the lowest threshold the accelerometer supports.

commit a40ad665 (2017-07-10): Bugfix sensitivity options typo + add more options

Changes sensitivity scale to be more precise.
Fixes #17
closed 2017-08-03 by Ralim