Ralim/IronOS #21: Switching buttons for cycling through settings/changing values

twikis, 2017-07-11:
So you're entering the settings with the button near the display and you're cycling through the settings with the button closer to the tip. I think it'd be more intuitive to use the button you're using to enter the settings to also cycle through the menu (it quite regularly happens to me that I'm changing the UVCO when I want actually want to go to the next setting...).

Any reason against remapping the buttons?

kokein, 2017-07-12:
Same here! +1 to make it reversed.
Nick

Ralim, 2017-07-12:
Yep, this is fairly easy to do. 
Originally I picked this mapping so that each button corresponded with the side of the screen it would change (in right-handed mode).

Do you think it would be worth adding an option to flip the buttons, or just reverse them permanently?

twikis, 2017-07-12:
Now that you say it it sounds logical, but I would have never come upon the idea that this is the reason....

I don't think a menu entry is necessary.

Ralim, 2017-07-12:
Yeah, it seemed logical at the time when trying to pick a fairly arbitrary layout really.
That said I'm not overly opposed to either layout :)
I've pushed up a new build of the 1.13 software that has these buttons flipped so you can try it out.

twikis, 2017-07-12:
Tried out the reversed buttons, took me 2 tries to get used to it and now it's good. Thanks!

closed 2017-07-12 by Ralim