Ralim/IronOS #22: Adding a function when long pressing the button that changes values in the settings

twikis, 2017-07-11:
I think it'd be great to have a function when you long press the value-change-button, e.g. changing the value backwards or having it do 2-3 steps a second while holding the button.

Also, how about immediately exiting the settings menu when holding down the "change settings" button for 1 second?

Any better ideas?

Ralim, 2017-07-12:
The only issue here is detecting long button presses hasn't been coded yet, and as of current I probably don't have the time to implement it just yet.

twikis, 2017-07-12:
Well, here is another idea:
The iron is missing a third button to switch between edit mode and cycle mode, right? When in the settings menu, how about turning the iron ~90 degrees (around the "screwdriver" axis) and then turning it back to enter "edit mode", changing values with button A (to decrease the selected value) and button B (to increase the value), then doing the same short "flip" movement to go back and cycle back and forth through settings (also with button A and B).

When doing it like this, you could also exit the menu by just clicking both buttons. And something I'd really appreciate is a function that remembers the last setting used (e.g. boost mode temperatur).

What do you think?

Ralim, 2017-07-12:
I would rather not require the use of movement of the iron, as sometimes it's very hard for some users of the iron in some situations to move the iron in arbitrary movements just to change settings.

I will probably look into long press detection first as that's the easiest solution to implement for now, though the option is always available to come back to.

Exiting the menu by both buttons is logical, and I can definitely add that :)
Jumping to the last used menu item would then require only having both buttons pressed as an exit function. This is not an issue, just more change on the already existing users of the firmware.

Why are you changing the boost temperature soo much?

gordio, 2017-07-12:
> I would rather not require the use of movement of the iron…
👍 Buttons are more intuitive.

twikis, 2017-07-15:
How about shaking? ... Alright I lost. :D

Thank you for creating the long press function! In my opinion it could be 50-100% faster though, as I can click the buttons a lot faster than what the long press function does... How about a menu entry to change the speed?
Also, when I push the buttons quickly it sometimes "misses" my clicks, almost as if the speed it limited to what can be reached via the long press function.

Actually I played around with the "both buttons thing", and I guess it may be possible to change a settings value when trying to exit, which would be annoying... What works much better for me is to just long-press button B until it exits the menu... :)
(But it also enters it again when you don't release button B quick enough, is there an easy fix possible here?)

Not that often actually, but still the setting I'm changing the most.

Ralim, 2017-07-16:
Motion control is just a pain unless the device is designed for it (ES120 style).

I can speed it up, however having the repeat too fast can be harder for users who do not use the iron/menu often, and they lose the *touch* of rapid buttons. The button speed is limited, as this was part of the simple and easy method of debouncing the incoming button press signals. Buttons must be stable for 30mS+ before they are registered.

It's not a hard fix, but I prefer not to have special case button behaviour, it's much nicer on the user to have all buttons behave identically.

closed 2017-07-31 by Ralim