Ralim/IronOS #23: i get a constant ERR code in the HEX file transfer

Jigal007, 2017-07-13:
i've got the newest TS100 with DFU 3:43
I have Windows 7, 64 bits and when I drag the .hex file (Flash Backup.hex) onto the USB drive, then
after rebooting it changes in ERR. then updating to version 1.27 (original hex-version) works OK.

What am I doing wrong?

Ralim, 2017-07-13:
Hi, 
Please check the instructions on the main page, you need to use the latest .hex file from the releases page. That file is called do not flash as it's only for developers who are directly messing with the units insides.

Jigal007, 2017-07-13:
Hi Ben,

Thank you for your response. I tried that file (version 1.13) also but I get "DONOTF~1.ERR" as a result.
I tried the same with (original) version 2.12 and then 2.17 (latest) and all went OK.
Maybe there are different versions of TS-100, I bought mine on April 28th this year.

Jigal007, 2017-07-13:
I used the hex-file from the ZIP-file, is that OK?

Jigal007, 2017-07-13:
Stupid mistake! I used the file from the ZIP-file, instead of the hex file on the release page. I use Firefox with NoScript and when I download that hex-file I get an xml-file instead (a0660986-6750-11e7-8be6-d8fab750c37b.xml). I downloaded it with good old Internet Explorer and all went OK.  Thanks for helping me.

Ralim, 2017-07-13:
No problem,
The releases system is a much neater way to handle software releases as old versions do not pollute the repository (and means you don't need to download the entire zip each time). 
This is a bit different to some other software out there though, so it just takes a bit to get used to the first time you see it :)

closed 2017-07-13 by Ralim