Ralim/IronOS #24: USB-C?

daling, 2017-07-13:
Hey, a few weeks ago I got a pretty awesome idea: would it be possible to hack this iron into having USB-C PD input? 

The USB-C specs allows for 3A at 20V, or even 5A. This would mean that it would be able to power the iron from like any common available charger. Sadly I don't have the knowledge nor the skills to pull this off . 

Wouldn't that be pretty awesome? :D

I did some research and fount out that the [STM32F0 devices can be used for USB-C PD](http://www.st.com/content/ccc/resource/technical/document/user_manual/group0/76/14/07/60/cc/3b/46/21/DM00288897/files/DM00288897.pdf/jcr:content/translations/en.DM00288897.pdf). The STM32 in this TS100 is a STM32F1 and seems currently not supported. But my knowledge about STM32's and USB PD is very limited, maybe someone out there is able to tell us more.

joric, 2017-07-14:
USB-C PD is not that intelligent. It simply reacts to data line levels:

| D+        | D-           | VCC  |
| ------------- |:-------------:| -----:|
| 0.6 | 0.0  | 5 |
| 3.3 | 0.6  |   9 |
| 0.6 | 3.3  |    12 |
| 3.3 | 3.3  |    20 |

So 20 volts is actually very simple to get, except there are very few Class B (20v capable) devices on the market, most of them are 12v max.

Sadly USB Type-C is much larger than microusb and probably won't fit inside.

For starters you can get 12V from the QC-capable power bank using [QC2.0 3.0 Quick Charge Adapter](https://www.aliexpress.com/item/1PCS-QC2-0-qc3-0-Tester-Automatically-Detects-Phone-Quick-Charge-Voltage-Meter-Simulator-Trigger-9V/32771660912.html).

It actually works:

[image]

[image]

daling, 2017-07-14:
Wow, I expected it to be a lot less simple.

So if I understand this Quick Charge Adapter correctly, the only thing it is doing is giving different voltages over the data lines. That's something the STM32 should be able to do as well.

I can see that the USB-C connector is too big, but we can make USB-C to DC5525 + MicroUSB adapter cables. Or MicroUSB to DC5525 + MicroUSB for QC3.0 devices.

Here's my actual feature request:
Add a mode where it sends max. voltage (3.3) over D+ and D- on the MicroUSB connector.

Ralim, 2017-07-14:
Hi,
You are talking about two different things here: QC2.0 chargers and USB-C.
QC2.0 uses the data pins to signal higher voltages by looking at what value is present on the data pins.
USB-C requires the device to identify itself and actually ask for the higher voltages.
Working with QC2.0 chargers can be done entirely without the microcontroller using any of the available intermediate boards or by setting up a voltage regulator and resistors inline to the unit (since you would need to make a custom power cable anyway).

USB-C is completely out of the question for the STM32f1 inside to implement as the pins are just not routed in a way to make it easy to do. It would be far easier to use an external module to perform the negotiation to ask the power adapter to supply the max voltage. This could be done in a similar fashion to [this article](https://www.reclaimerlabs.com/blog/2017/3/14/usb-c-easy-bake-oven).

perillamint, 2017-07-14:
Someone did it using Hakko handle and custom controller board.

[http://hackaday.com/2017/02/24/60-watt-usb-soldering-iron-does-it-with-type-c/](http://hackaday.com/2017/02/24/60-watt-usb-soldering-iron-does-it-with-type-c/)

I think this setup also works for TS100 iron.

Ralim, 2017-07-14:
I think @perillamint is on the right track.
Using an external converter board that also houses the USB-C connector would make the most sense given the already compact design of the ts100 iron. Would also allow you to power other devices with the same converter board.

gordio, 2017-07-22:
@Ralim a small improvement (board with QC USB) and USB-C connectors :-D

Ralim, 2017-07-23:
I'm currently a little bit too busy to design a dedicated pcb at this point in time. But ill come back to this when im less busy to see where the market is at for this kind of option :)

qoobaa, 2017-11-21:
I was looking for the simplest way to power TS-100 or charge my laptop out of a USB PD enabled charger, and I've just stumbled upon this guy: https://www.aliexpress.com/item/USB-PD-trap-TYPE-C-Fast-Flip-Flop-pollinpg-device-detector-trigger/32830872838.html. I've just got one and it seems to work flawlessly.

joric, 2017-12-04:
@qoobaa yeah cool but one I was referring to (and already posted a link above) costs just $3, not $13 https://www.aliexpress.com/item/1PCS-QC2-0-qc3-0-Tester-Automatically-Detects-Phone-Quick-Charge-Voltage-Meter-Simulator-Trigger-9V/32771660912.html

qoobaa, 2017-12-04:
@joric I bought this one too, but it's not USB-C PD.

joric, 2017-12-04:
@qoobaa ah ok it's quickcharge. Different standards.

RRAlex, 2018-02-04:
It'd be nice if there was a simple USB-C PD (20V) -> 5.5mm dongle adapter.
There are [some batteries](https://www.ravpower.com/26800mah-Type-C-external-battery-charger-black.html) now that support 30W output, which is probably enough for many jobs?

gordio, 2018-02-04:
It will be nice to put this port in TS100 - [USB-C Port](https://user-images.githubusercontent.com/141126/35781892-9b3f6060-09f8-11e8-8c14-b5493cda4c95.png), and use Power Bank like this (12V/1.5A) - [Tronsmart-Presto USB-C](https://www.aliexpress.com/item/Tronsmart-Presto-10400mAh-PBT12-Power-Bank-Quick-Charge-3-0-USB-Type-C-Powerbank-External-Portable/32833489732.html)

clowrey, 2018-02-17:
I would love to see USB C implementation directly on the next TS 100 - it is the elegant solution. Data + power all in one and many new laptops support this connector.

Ralim, 2018-02-26:
The next generation (TS200) will use Quick Charge 2.0 over USB-C. Sadly it looks to be limited in power to a low power (~20W). Specs haven't been confirmed just yet though. Just looking at marketing material.

I'm personally going to stick to the TS100 for now. But will update here if I hear any more information.

JohnEdwa, 2018-02-28:
Why use QC2 (which has a maximum output of 20W), when USB-PD could provide all the way up to a 100W. Guess they just thought there are more QC2 compatible chargers around?

Ralim, 2018-02-28:
Im guessing so? Havent really heard the reasoning, but 20W is just too little power for most things if you ask me. PD would be much nicer.

I dont have any links online im afraid :(

Adrian-at-CrimsonAuzre, 2018-03-01:
@qoobaa Does that Ali board automatically try to pull the max voltage from USB-PD devices? I'm thinking about getting one and a DC5525 to make my own USB C adapter.

qoobaa, 2018-03-01:
@Adrian-at-CrimsonAuzre it does not. It requires to push the button several times after connecting, sometimes it doesn't allow to choose the highest profile (randomly). I gave it up and ordered PD Buddy from Tindie - it works perfectly, but it's quite pricey compared to the Ali board. No idea about passing the data lines (I'd bet it does not pass anything).

Adrian-at-CrimsonAuzre, 2018-03-01:
Might end up getting one of those or making my own.
If I can cram the circuitry into a small enough board I'd probably replace the 5.5mm plug with a daughter board.

alexnif007, 2018-03-10:
@Ralim IMHO, whatever the name of the technology is (PD or Quick Charge 2.0-3.0), everyone here wish to get possibility to make some smart battery pack, supporting higher voltage and currencies to output at least 12V with adding minimal additional external hardware for TS100. Not adding USB-Type C (it is useless, and will not give a required result). Here a man did something same with a few lines of code for ATTiny85: http://blog.rnix.de/12v-from-a-usb-powerbank/   The question is: Is it possible to add a few lines in the firmware of TS100, so, that it will be capable of giving corresponding signals through it's microUSB, for instance, selecting some menuitem or just with a press of a key, like it is described in the link, given before? Or not?

JohnEdwa, 2018-03-11:
@alexnif007 
You need the resistors in the datalines to be able to pull them to 3.3V for the QC identification (which I'm pretty sure the TS100 lacks), so you would at the least need to make a special cable for it.

Ralim, 2018-03-11:
@alexnif007 
I wouldn't recommend anyone designing for QC 2/3 as its being strongly pushed out in favour of USB-PD which is available on USB-C. As well as USB-PD supporting higher voltages and higher power. 

However all of these require some form of negotiation, either resistors(QC) or active messages(USB-PD).
For all of these you are going to need to make up a special cable of some kind to :

1. Connect the power output through to the barrel jack (you cannot supply heater from USB)
2. Interface to the standard using resistors / communications
3. Adapt to plug in the case of USB-PD

It would be possible to put some of this inside the iron, sure, but the vast majority needs to be added externally which is why most people just purchase adapters and call it done.

I would happily support someone making a USB-C to barrel jack USB-PD cable which would be awesome, but is not really something that can be done in the iron itself.

The 20W limit on QC is a joke for a soldering iron that is going to be used for anything large. 
20W is workable for low powered use (small assembly), but as soon as you try and solder a larger connection you really want the higher power available in USB-PD.

alexnif007, 2018-03-11:
@Ralim While the months pass - still on the market I see only QC chargers, so while here discussion continues already half a year - it started 17 Jul 2017, this half year already people could use it "on the road" with the pocket battery. The TS100 itself is - not serious - it's handle getting extremely hot in a few minutes, and it has no secure stop for a hand not to slide to the heating element, so it cannot be used for something large whatsoever, but only few pinches "in the field", for which is needed portability, also of a battery, which is being in use, in this case mostly present QC on the market, if take something more powerful - TS100 losing it's main advantage - PORTABILITY, because of giantic battery, which someone has to carry to use it. For "anything large" I prefer to use butane Dremel or portasol - offering the same portability with more power on the field. Anyway, I am going to do QC bit bang with Arduino which  power the TS100 using barrel, for connection of Arduino to the battery use USB. Just before didn't want another external element adding clumsines to the TS100 construction.

perillamint, 2018-03-13:
IMO, for this kind of application, USB-PD is more suitable then QC2 or 3(reverse engineered -- http://blog.deconinck.info/post/2017/08/09/Turning-a-Quick-Charge-3.0-charger-into-a-variable-voltage-power-supply). QC 2 or 3  only provide power up to 18 Watt, which doesn't enough for this kind of application.

In contrast, some popular USB-PD adapters like [Apple USB-PD power brick](https://www.apple.com/shop/product/MJ262LL/A/29w-usb‑c-power-adapter) starts from 29W and [Nintendo Switch power brick](https://www.reddit.com/r/NintendoSwitch/comments/5l30zb/switch_has_39w_ac_adapter_fcc/) provides up to 39 Watt, which will definitely do its job.

alexnif007, 2018-03-13:
@perillamint If talking about usage of 110V or 220V power adapter - there's no need to make it this complicated way, you can choose any even more powerful notebook charger, even 3.75A or 4A, just select needed barrel size from wide selection of chargers for some Dell, ASUS, Acer notebooks, it will be even cheaper. Here we are talking about usage of some POCKET BATTERY!

Ralim, 2018-03-13:
@perillamint I agree that QC is not worth touching, and the growing market of the usb pd is worth working towards using as it works well for portable battery supplies as well as wall adapters that are available. 

What would be nice is to create an add on that decodes the usb-c pd and communicates to the iron the available power so it can scale to the available power. 

I agree @alexnif007 about this being a discussion about portable batteries, but it's important to also look at the wider use case of people who use their ts100's at a desk and like to use mains power.

JohnEdwa, 2018-03-13:
I use a [tiny 19V @ 2.1A (40W) laptop charger](https://i.imgur.com/yCyUHFw.jpg?1), it has the perfect balance of size and power for me, and the lower power means the cable is thin and extremely flexible. Just slightly ruined by the gigantic EU plug (I've since swapped the travel adapter plug to a regular 0.5m cable), but there's not much I can do about that :)   
If I really need to take it to the field, a 4S lipo works fine.

@Ralim 
> What would be nice is to create an add on that decodes the usb-c pd and communicates to the iron the available power so it can scale to the available power.

AFAIK implementing proper USB-PD is not easy to DIY, but there are plenty of dirt cheap manager chips one could use, like the [FUSB302](http://www.onsemi.com/PowerSolutions/product.do?id=FUSB302) that could be used (the chip is ~$0.35 in quantity). [Here's a project/breakout board](https://www.tindie.com/products/ReclaimerLabs/usb-type-c-power-delivery-phy-breakout-board/) that can basically do what the second post QC adapter, but for USB-PD.  
I guess why no-one has really done anything like this is the danger of having a regular USB port that is configured to output 20V@5A to anything you plug in it, which is why I'd much rather just skip that part and just make a device with a USB3 plug in one end, the USB-PD chip in the middle with a switch to select between a few voltages, and a 5.5mm barrel jack in the other.

Ralim, 2018-03-14:
I would only do usb-pd using an interface ic such as the one you mentioned. Your proposed is basically what I was thinking.

I think either an inline cable that provides a barrel output or a new iron entirely are the two ways we are most likely to see this.

clowrey, 2018-03-14:
+1 to USB PD built into the iron... (a new model) imagine just connecting your USB C PD adapter straight from laptop to the iron.. Ideal solution.

moracabanas, 2018-03-29:
I Bought this https://www.aliexpress.com/item/USB-PD-trap-TYPE-C-Fast-Flip-Flop-pollinpg-device-detector-trigger/32830872838.html and I can confirm that:
1. It works with a USB-C PD 40W ZMI (Xiaomi) 20000mah powerbank just as many of you wanted.

2. It actually switches automatically to the highest Voltage the USB-C PD can delivery. So if you plug it then it takes 20V right the way. And then, you can push the button to switch to another voltage.

I bet for the combo and It is just the best investment i couldn't be done in years.

Adrian-at-CrimsonAuzre, 2018-03-29:
@moracabanas I'm guessing that when you press the button, it goes to Level 1>2>3>...?
I was going to make a board for this (PD Buddy is a bit bulky), but this will work perfectly.

qoobaa, 2018-03-30:
I also bought that "USB PD trap". My version definitely didn't switch to the highest voltage automatically, sometimes prevented to choose the highest voltage profile manually too (possibly a bug) - so I had to disconnect it and connect it again to do so. I replaced it with PD Buddy board eventually - it works flawlessly now.

joric, 2018-03-31:
@Ralim yeah there is not so much information about TS200. Are there any pics? I only got a few from the forum here https://mysku.ru/blog/china-stores/60248.html they say it's from some chinese chat. Any other source?

https://pic.mysku-st.ru/uploads/pictures/06/84/74/2018/02/03/72db62.jpg

https://pic.mysku-st.ru/uploads/pictures/06/84/74/2018/02/03/63f5cc.jpg

KarmusDK, 2018-04-07:
@Ralim I bought this cable once: https://www.tabletcovers.dk/products/macbook-usb-c-til-dc-5-5-mm-forlaenger-opladerkabel-2-m-sort
Is it possible to use this cable to pull 12V or more from a PSU or powerbank with QuickCharge or Power Delivery?

JohnEdwa, 2018-04-08:
@KarmusDK Not really, as there are no active electronics in the cable to tell the PSU how much power it wants, and the iron has no data connections to the barrel jack (all it can do is turn it on or off through a mosfet).

But if you get a device to output enough power out of a USB-C port without active communication from the iron (like using one of the QC or USB-PD spoofing boards discussed in this thread), then sure.

KarmusDK, 2018-04-08:
@JohnEdwa Thanks for replying so fast. I looked a bit into different trigger boards, although they all seem to have the same connector standards: USB-A and USB-A or USB-A and none (holes for wire soldering). However, I stumbled across [this advanced measuring tool](https://www.unionrepair.com/how-to-use-power-z-usb-pd-tester-voltage-current-type-c-meter-km001/?currency=USD&sl=en). It seems to have trigger functions as well, and even a USB-C OUT port. Does anyone have experience with this module? It would be expensive to buy, just to find out that it cannot be applied to my instance.

Ralim, 2018-04-13:
@KarmusDK I don't think that module will do what you want I'm afraid.

I have the @ReclaimerLabs breakout board on my desk to play with. It combined with a microcontroller would do what you want, but stand alone it wouldnt really do much else. @ReclaimerLabs does make a nice board that has a microcontroller combined with it that could be programmed to automatically request the max available and that would do what you wanted. But still a pricy option :/

KarmusDK, 2018-04-27:
It works! The module has options to choose from and apparently is able to manipulate both QC2.0, QC3.0, USB PD and various other protocols. I succeeded in running a [UDOO X86 Ultra-microcomputer](https://www.udoo.org/udoo-x86/) off a [QuickCharge-capable](https://www.razer.com/gaming-accessories/razer-power-bank) (unsure about Power Delivery though, but apparently it's a feature) [power bank](https://www.razer.com/gaming-accessories/razer-power-bank) with both USB-A and USB-C outputs with the 12V/3A profile on a 2.1mm/5.5mm DC barrel jack. I have yet to test it with the TS-100 and its 2.5mm/5.5mm, but since this is also an analog device, I expect a great possibility of instant success.

[image]

Adrian-at-CrimsonAuzre, 2018-04-30:
So I bought https://www.aliexpress.com/item/USB-PD-trap-TYPE-C-Fast-Flip-Flop-pollinpg-device-detector-trigger/32830872838.html and it will not select the highest profile. It will also jump down to 5V after a few minutes of use, turning off the iron.

KarmusDK, 2018-05-05:
The TS-100 soldering iron works with a USB-C to DC 2.5mm/5.5mm plug and the Power-Z KM-001 module's on either QC3.0 (QuickCharge) or USB PD (Power Delivery) test functionality. It heats up quite fast on 20V.

[image]

[image]

It is absolutely crucial to have a portable soldering iron with an attached battery bank if you're a technician on the go, but be sure to buy one with an immense capacity - alternatively you can go for a gas-powered iron.

RRAlex, 2018-05-07:
Great work, thanks for confirming this!

Where did you find the USB to DC adapter?

KarmusDK, 2018-05-07:
> Great work, thanks for confirming this!
>
> Where did you find the USB to DC adapter?

https://www.tabletcovers.dk/products/macbook-usb-c-til-dc-5-5-mm-forlaenger-opladerkabel-2-m-sort

RRAlex, 2018-05-27:
Not quite USB-C, but for the record if you are into FPV drones and have spare (4S?) batteries with XT60 connectors, you could [try this cheap adapter](
https://www.banggood.com/XT60-Male-Bullet-Connector-to-Male-DC-DC5525-Power-Cable-for-TS100-Electronic-Soldering-Iron-p-1219960.html).

Adrian-at-CrimsonAuzre, 2018-05-27:
I actually bought this adapter and modified it to use with the PD Buddy.
It's a really nice and flexible silicone lead.

joric, 2018-08-04:
TS80 is selling now on banggood and aliexpress for $72.99.

xavkno, 2018-08-04:
Shame it isn't USB PD compatible but Qc 3.0 instead.

joric, 2018-09-22:
@xavkno yep. Doesn't power on with 65W laptop PD adapter (CDQ07ZM) https://youtu.be/xc-yIq0m54o

milpugeda, 2018-10-17:
Any updates regarding this? So far it looks like:

@qoobaa and @Adrian-at-CrimsonAuzre has tried this but is having issues with selecting the highest profile so it's no good:
https://www.aliexpress.com/item/USB-PD-trap-TYPE-C-Fast-Flip-Flop-polling-device-detector-trigger/32830872838.html

PD Buddy works but is pricey
https://www.tindie.com/products/clayghobbs/pd-buddy-sink/

Also the Power-Z can trigger PD but is also pricey
https://www.banggood.com/POWER-Z-PD3_0-QC4_0-USB-Tester-Voltage-Current-Tester-Ripple-Dual-Type-C-Meter-p-1226355.html

The other boards I have seen are these:

This supports both QC2.0 and PD https://www.aliexpress.com/item/2D001-PD-QC2-0-trap-trigger-polling-device-low-internal-resistance-hardware-scheme/32816201591.html

https://www.aliexpress.com/item/ZY12PDS-Type-C-PD-to-DC-USB-SurfacePro-Decoy-Fast-Charge-Trigger-Polling-Device/32911057550.html

Anyone else tried something that works but is priced a bit cheaper? What's the best way to use PD on the TS100?

Adrian-at-CrimsonAuzre, 2018-10-17:
The PD buddy and [one of these](https://www.banggood.com/XT60-Male-Bullet-Connector-to-Male-DC-DC5525-Power-Cable-for-TS100-Electronic-Soldering-Iron-p-1219960.html?cur_warehouse=CN) from wherever you choose to get it is the best option if you want USB C. It's all but guaranteed to work, and it's easy to solder together and setup.

milpugeda, 2018-10-17:
@Adrian-at-CrimsonAuzre You mean both can be used for portable power? Because the XT60 connector is not USB C right? Just something to connect a Lipo battery.

Adrian-at-CrimsonAuzre, 2018-10-17:
@milpugeda You chop the battery end off the connector and solder it to the USB C board. Then configure the board to request the max power from the USB C source

qoobaa, 2018-10-18:
Take a look at ZY12PDS on Aliexpress. It's much better than the previous board. I used it to make an adapter for Thinkpad, and it works flawlessly. It costs $12, allows to select voltage by soldering pads on the PCB (no buttons). It's also very tiny - can be easily wrapped in using a heat shrink tube.

milptp, 2018-10-18:
@qoobaa Hey that looks awesome! How do you choose the voltage? I can't seem to understand the instructions.

[image]

Like how do I set it to 20v for the TS100?

I'm also thinking of getting the version with the USB-A port and the button so I can cycle thru the voltages and just use a USB to DC cable. By the way, does the ZY12PDS support both PD and QC3.0?

milptp, 2018-10-18:
@Adrian-at-CrimsonAuzre Hi man, I'm the same user you replied to. Is it okay if you delete your previous comment with the @ mention to my other account? I mistakenly used my other account which I prefer not use in public. Sorry bout that, sucks that Github doesn't have private messaging.

qoobaa, 2018-10-18:
It's 20V by default, (XX means no connections). If I understand correctly, you just need to solder these pads according to the description below them (I haven't tried), X means no connection, O means the pads are soldered. 

[image]

milptp, 2018-10-18:
So for your cable you didn't solder any of the pads and your using 20v? This might be the cheapest way to have USB PD on the TS100. Based on the product description ZY12PDS might support QC3.0 as well but I can't be certain because the translation is pretty bad. The other PCB versions that have a button also have an LED indicator that can change color to indicate which voltage is set.

qoobaa, 2018-10-18:
Yes - it's 20V without soldering any pads. I use it with an Anker power adapter with PD support *only* (no QC), works perfectly.

clowrey, 2018-10-18:
That is awesome! I bought the earlier model from YZXStudio - but it required pushing the little button to switch between the voltages, and if wired directly to the TS100 it will enter into USB programming mode when powering up on 5V - so it never goes to soldering mode even after 20V... But this model that does 20V always hopefully avoids this problem? Has anyone tested yet?

qoobaa, 2018-10-18:
I've got two spare modules atm, I'll test it when I find some time.

[71 more replies]

closed 2023-06-05 by Ralim