Ralim/IronOS #27: Text position when flipped

Nredor, 2017-07-16:
[image]

Picture shows it all. Text position is bad when display is flipped.

Ralim, 2017-07-16:
Woah that's not meant to happen.
I can't reproduce the issue on my Irons here, any chance you can tell me what version of the firmware you are on? Also, where did you buy your iron though?

Is it only on the soldering screen, or is it on all screens?

Nredor, 2017-07-19:
Tried now with latest version. Ordered from https://www.banggood.com/MINI-TS100-Digital-OLED-Programable-Interface-DC-5525-Soldering-Iron-Station-Built-in-STM32-Chip-p-984214.html

Text is displayed fully but few pixels of the display are behind black mask. You see the whole text if you look from an angle.

Nredor, 2017-07-19:
Ok, now that i look with bright light seems that my units display is not positioned correctly. So I probably have faulty unit ;) Software seems to be working as intended.

closed 2017-07-19 by Nredor

Ralim, 2017-07-20:
It should be really easy to align the screen if this is just the issue. 
I might add a little bit more padding on the screen when I have time to test it too.