Ralim/IronOS #28: USB: Device can't be enumerated under linux when running the Ralim fw

doegox, 2017-07-18:
Hi

When running Ralim (1.13) and plugging under my Debian (kernel 4.9.0), it fails enumerating the device:

```
Jul 18 14:12:53 mercure kernel: [133910.311538] usb 1-1: new full-speed USB device number 6 using xhci_hcd
Jul 18 14:12:53 mercure kernel: [133910.431488] usb 1-1: device descriptor read/64, error -71
Jul 18 14:12:53 mercure kernel: [133910.663480] usb 1-1: device descriptor read/64, error -71
Jul 18 14:12:53 mercure kernel: [133910.891403] usb 1-1: new full-speed USB device number 7 using xhci_hcd
Jul 18 14:12:53 mercure kernel: [133911.011407] usb 1-1: device descriptor read/64, error -71
Jul 18 14:12:54 mercure kernel: [133911.239573] usb 1-1: device descriptor read/64, error -71
Jul 18 14:12:54 mercure kernel: [133911.467345] usb 1-1: new full-speed USB device number 8 using xhci_hcd
Jul 18 14:12:54 mercure kernel: [133911.467476] usb 1-1: Device not responding to setup address.
Jul 18 14:12:54 mercure kernel: [133911.675711] usb 1-1: Device not responding to setup address.
Jul 18 14:12:54 mercure kernel: [133911.883431] usb 1-1: device not accepting address 8, error -71
Jul 18 14:12:54 mercure kernel: [133912.003443] usb 1-1: new full-speed USB device number 9 using xhci_hcd
Jul 18 14:12:54 mercure kernel: [133912.003649] usb 1-1: Device not responding to setup address.
Jul 18 14:12:55 mercure kernel: [133912.215747] usb 1-1: Device not responding to setup address.
Jul 18 14:12:55 mercure kernel: [133912.423512] usb 1-1: device not accepting address 9, error -71
Jul 18 14:12:55 mercure kernel: [133912.423584] usb usb1-port1: unable to enumerate USB device

```

I swapped several times with the official firmware (2.17) and this is reproducible. The DFU and official firmware (config) modes get both enumerated properly and device can be mounted, but it fails completely with Ralim.

Ralim, 2017-07-18:
This firmware does not implement USB at the moment, this is expected behaviour. The next release will hold the D+ line low so it won't complain that it's not responding (instead it will just not enumerate at all). 

DFU mode still uses USB, but it's entirely separate, as we still use the original DFU for updating the code.

doegox, 2017-07-18:
ha ok, yes would be cleared to avoid completely any attempt of enumeration, thanks for the explanation!

closed 2017-07-18 by doegox

Ralim, 2017-07-18:
No problem 😃 
I should probably put something more on the main page to mention this...
I dropped USB as it couldn't be compiled on open-source compilers easily, and the menu replaced its functionality.