Ralim/IronOS #29: Left hand Display

ltwin8, 2017-07-21:
The intro icon wich shows the iron with the buttons is the wrong direction. also the time to press both buttons are very short to exit solder mode...

Also very useful would be a permanent charge indicator for the battery, because most people I think use it with lipos and there would be nice to have for example a bar of pixels permanently on the display wich length relates to charge.... only 3 more menu items: CLOW (CELL LOW) GHIGH (CELL HIGH) CCOUNT (CELL COUNT)...

Great firmware for a rewrite until now :) 👍

Ralim, 2017-07-21:
Hi,
I agree the left handed mode shows the icon reversed to what would be logical. I haven't had the time to sit down and draw in the flipped image as of yet sadly.

The intention is for pressing both buttons to be extremely hard to do accidentally, and require two hands (or a large thumb) to do correctly. Is this an issue for use of the iron? It is possible to add an extra delay to slow down the response time of the buttons, however, currently, the system is coded such that the response time of buttons anywhere in the use of the iron is the same.

I like the idea of a battery display, however, a large number of users of these irons use them on a non-battery supply. So requiring them to skip past multiple settings can be frustrating.

That said, most users use standard Lithium polymer batteries. If users are willing to live with set thresholds of like, minimum being > 3.V and cell maximum being >4.1V, then it would be possible to have the unit automatically detect the cell count and only require a setting to toggle the display of the battery status on and off?

Would that work acceptably for you?

ltwin8, 2017-07-21:
Hello, thanks for the fast respond.

The delay for the two buttons would be great.

I see a lot reviews of people they use it for repairing for example copters and use simply their lipos to power. Automatic cell count is a bit complicate, for example: an nearly empty "normal lipo" 6S can have 21V (3,5V =~20-30%) can have the same voltage as a full 5S „normal lipo“ at 4,2V per cell… and also there are LiHV lipos (4,35V). I wouldn’t toggle the display I think a little bar at the bottom or side of display would be enough, or toggle when „shake“ the iron…. Maybe a warning when the cell-voltage reaches a set value the display shows an alarm? 
That would be absolute awesome features.

Best regards and keep on doing this great work :)

Ralim, 2017-07-21:
I like to try and respond quickly.

I'll look into making the detection a bit slower so its easier to hit the buttons. Might be a bit difficult with how the code is currently structured.

Yeah, that's a fair point about HV cells, completely mind blanked on them. 
I'm really trying to avoid adding too many extra menu items, as it's already getting very long with options. I could add a menu option for "power source" that selects between 3-5S and DC so that it can automatically configure if it shows the battery status or not?

ltwin8, 2017-07-21:
Hello,

The idea with the extra menu is great, but I would make the option to go even to 6S (25,2V) it is not within the specs but it works because the swithingregulator can take it and the voltagedevider is a bit tricky but the ADC input from the mcu can take 3.6V regarding to data sheet. Wich would theoretic allow 25,5V. I can’t say if the soldering tip handle 25,2V but it should, maybe the message on release page: „6S only with non LIHV LiPos and on your own risk 6S LIHV would be destroy your Irons processor“.

A big plus of this would be that the iron than take all between 3S and 6S. And it would protect the battery.

What about a point in the menu „PWR“ with settings „SMPS“ and „BATT“ where batt adds the ability to make the settings for the voltage warning? I don’t know why you need low voltage alarm on a fixed supply because it would always deliver the same voltage.

Best regards

Ralim, 2017-07-21:
The input filter capacitors inside the iron are only rated to 25V input.
So you are technically overvolting the input capacitors. Also running the iron at the higher voltages also affects the voltage measurement as the overvolting of the ADC input does appear to degrade the input electronics.

While I can add the option for 6S, I don't want to encourage anyone to damage their irons.
So I will need to add a warning to the main page since exceeding the 25V input degrades the filtering capacitors and also affects the accuracy of the ADC readings.

I was thinking of using a combined menu, but the problem is for people who are using non-standard input sources, where the cutoff voltage may not align with normal LiPo cell voltages?

ltwin8, 2017-07-21:
hello,

The ADC should be able to convert that,
 „
2 x 12-bit, 1 μs A/D converters (up to 16 channels)

–  Conversion range: 0 to 3.6 V 

„

The capacitors are only 25V? Okay that is bad, normal you take some like 35V for 24v, normally elkos so close to the maximum rating is not so good (voltagespikes).
I agree, nobody should damage their irons, thats not so good.

The problem with the cutoff voltage is right, maybe a further combined menu where an option Batt and Lipo are in? Because the one point cutoff voltage is than only taken for alternating sources like car battery or so… 

I like your idea with combined menu, it takes the iron simple and also adds the functions for protect batterys if used, so that users can Cutoff the power more exact (like 10V for a 3S is very low while 11V is quite high...)

Best regards

reversebias, 2017-07-23:
Even though the caps are rated for 25V on the schematic, it's only the input filtering cap for the 3.9V switchmode converter. There's D4 (1N4148) and R30 (150R) in series with the supply to the caps, which drop about 2.5V under normal conditions. I measured 22.6V on the cap with a 25.2V input voltage. Although 25V caps are a bit marginal here, I don't think that using a 6S lipo is a safety risk.

Ralim, 2017-07-23:
Hi,
While I agree with you that you *should* be fine. I need to put the warning there, as its running the unit out-of-spec. This is why I added the 6S menu option, but I also added the warning to the main page.
Anway, can we please move the discussion on battery stuff over to #30 .

Ralim, 2017-07-24:
@ltwin8 Can you please clarify with the buttons to exist soldering mode, did you mean that it was too easy to exit soldering more, or that you would like it to be easier to get the double button press (ie exit easier)?

ltwin8, 2017-07-24:
Hello,

To exit easier would be fine so that you don’t always have to have both hands free for pressing buttons. 

Best regards

Ralim, 2017-07-24:
I can't really make it a single button action, but I will try and make the time it uses for detection of the double button press a bit longer. If your thumb is large enough it should be fairly easy to push both with one hand?

ltwin8, 2017-07-24:
yes I think with a bit training it would be easier to exit it with one hand.

maybe I found an other thing today: when I adjust temperature in soldering mode (Left hand) the temperature rises with the "usb-end" button, is that right?

Best regards

Ralim, 2017-07-24:
See how you go with it currently trying left handed mode. I'll try and extend the time slightly in the next version.

And yep that is a bug :/ Ill fix that too in 1.16

ltwin8, 2017-07-24:
Yes I try left hand mode, with right hand I can't solder.
its not a big problem, its like complain on a very high level.

by the way, the way the battery status is shown is great, it don't overlay important informations like temperature and soldering status and it is good readable outside

joric, 2017-07-26:
Hey, reversebias! Nice to see you here, I'm currently trying to build mitosis out of EasyEDA PCB's :)
As for this left hand feature I suggest using accelerometer for autorotation.
Current accelerometer API is poor but all you really need (in 2.17) is to check for a single bit, e.g. gY_value.Byte.hi>0x7f ? -1 : 1; set corresponding screen orientation and then call Init_Oled().
I've used it here https://github.com/joric/ts100tris/tree/autorotate
You may object that autorotation might be annoying but hey if it's good for phones is good for the soldering iron and people kind of expect that from the device that has a 3-axis accelerometer.

Ralim, 2017-07-26:
Hi, I agree that auto rotation could be a nice thing to add. 
I haven't fleshed out the full api for the accelerometer yet as I didn't need to :P Its trivial to add this as a feature, so I should be able to soon. 

I will probably update the setting to be 3 options, so you can lock the orientation or have it automatic.

commit 1cbcba92 (2017-07-27): Adding auto-rotation to the screen

Adds auto rotation support using the proper orientation detection. (Not
using raw values).
Should Fix and close #29
closed 2017-07-27 by Ralim