Ralim/IronOS #30: UVCO -> Undervoltage cut out level

Cpt-Huck, 2017-07-21:
Ability to change UVCO -> Undervoltage cut out level, settable in 0. 1 Volt increments from 10-24V. 
So if you use 4S LiPo it stops at 14.6 - 14.8 V and not 15 Volt..

Cpt-Huck, 2017-07-21:
Not really an issue.. But I'm new here and really like the Costum Firmware!

Ralim, 2017-07-22:
Hi @Cpt-Huck,
Over in #29 a suggestion came up to provide a method to select if the Iron is running off a DC supply or a lipo battery pack. As then it can be simplified down to selecting a cell count in the menu instead.

I proposed changing the cutoff menu to instead be called "PWRSRC", and have the following options {DC,3S,4S,5S,6S}. Where DC would disable any UVCO and then the S count would set the iron up for a cutoff voltage of 3.3*cell count. This would also enable the display of a battery charge indicator on the iron (only really approximate as estimating cell SOC by voltage only works with no load).

Do you think this would work better for you? (Rather than having to cycle through all the 0.1V steps when changing the cutoff voltage)

Cpt-Huck, 2017-07-22:
I've read through #29.. 

Looks like the TS100 only has 25 Elko which is quite low? So using 6S will burn it "maybe" 

So DC or Off would be the same.. But there's no other situation, that I know why someone should enable this when not using DC or LiPo.. 

I agree implementing the new "Undervoltage cut off" (out?!) feature have the option "DC 3S, 4S 5S (maybe 6S? Idk) sounds good

Cpt-Huck, 2017-07-22:
I meant "DC" is an useful option but it should not be called "off"

Cpt-Huck, 2017-07-22:
And also don't set the low voltage to 3.3 Volts per cell.. Ouch

Most FPV pilots land their quads at 3.6 to 3.7 Volts.. 

So setting this to 3.6 V would be better, because it wouldn't damage the cell 

..  cut off at 3.6 Volt is under load.. 
So if you disconnect the battery, the cell voltage would go back up to 3.65 - 3.7 V (due no load?) 

If you have any questions please let me know! (I tell you that, because I'm good at English but not perfect)

Cpt-Huck, 2017-07-22:
Summary

DC, 3S, 4S, 5S (maybe 6S?! if it works)

Lowest Cell voltage at 3.6 volts. 

Still don't get it what "PWRSRC" mean 😞 Power?! 

(damaged/old cells tend to cell - drift when theyre this low.. Example: 4S LiPo cell 1 = 3.6 V cell 2 = 3.6 V, cell 3 = 3.3 V cell 4 = 3.6 V... But this scenario is bad haha)

Ralim, 2017-07-22:
Hi,
I have already implemented a first pass at the PWRSRC (short for Power Source) option.
I have gone with DC rather than off, since the iron really doesn't function below 10V, so it's nicer to enforce people to use a nicer power supply.

There is a massive, massive difference between when you are using a battery pack under high load (quad) and low load (soldering). We are talking about a maximum current of around 3A here, which is generally < 3C discharge on the pack. Look at some low C discharge curves for batteries, when discharging at low currents going down to 3.3V is really common (Most batteries are rated down to 2.75V if < 1C). Racing batteries are often terminated earlier on the voltage scale because of the massive voltage sag under higher currents due to the internal resistance rise as you deplete the cell at the higher currents.

6S will work so long as your pack is not completely full. (ie <= 4V per cell). So it will work in a pinch but I really wouldn't recommend it unless it's your only option.

There are many, many other methods of protecting from cells that are unbalanced. And those packs should be monitored by the user using other means via the balance port. The low cutoff voltage is chosen as that would be reached under load. A cutoff of 3.3V under ~3A load is not going to damage any properly maintained, balanced and monitored battery packs.

Cpt-Huck, 2017-07-22:
Okay.. 3.3 Volt is not very much.. But I'm not a Profi regarding to LiPo battery's.. 

Does 1.14 contain DC, 3S, 4S, 5S (6S?!) 
? 

Thanks for this implementation!
Further testing and reporting soon..

Ralim, 2017-07-22:
That was what I added for 1.14,since I did like the idea and it was a fairly easy change. It does go up to 6s

DrStein99, 2017-07-22:
How about if I'm using 18650 battery cells, they have different cut-off from li-po cells.  A battery-pack cell configuration feature makes this tiny thing complicated.   Who wants to be confused with battery types and cell layouts - if they just plug in from a +12v car battery ?  What is wrong with configuring  setting for a low and / or high cut off value?

gordio, 2017-07-22:
> So if you use 4S LiPo it stops at 14.6 - 14.8 V and not 15 Volt..

Protect you battery, stop at 15V ;-)

EricJanVanDenBogaard, 2017-07-23:
So, if i'm correctly understanding the 1.14 software has a lipo cell count function and has presets stored for S3,S4,S5,S6, cut off values, what i understand, a lipo should not be discharged below 3.2 volt, and a Li-Ion not below 2.9 volt. could the software be made that it measures the internal resistance of the Lipo/Li-Ion accu ? or is the difference too little for that ?

Ralim, 2017-07-23:
@DrStein99 :
Li-Po and 18650's usually have extremely similar discharge curves, and this is why I chose 3.3V as a cutoff voltage, as this allows for ~95% energy extraction from both technologies.
The reason I preferred going for a cell count rather than a high and low value is that the vast majority of users are using Lithium based batteries, which have very similar discharge curves (Except for LiFePo4). If you are running off a car battery just set it to DC, as it will still stop at 10V which is fine for lead acid.

I'm trying to prevent making it a multiple option menu, since there are many quadcopter users who run it on 3S,4S or 5S depending whats nearby. If I make it require too many changes then they won't bother for most of the time, which completely removes the entire point of the cutoff voltage.

@EricJanVanDenBogaard :
Internal resistance isn't really easy to do without having a means to measure current. (Since there is a variation in the tip resistances, and its temperature dependent).  Most lithium polymer batteries that I have used in production have been rated to 3V or 2.75V as a minimum voltage. Can you link to a datasheet of these 3.3V batteries?
The difference is tiny, though we do happen to have a large load (the tip). So it would be possible to measure on batteries with a higher internal resistance (such as old racing packs, or 18650 packs). But on larger packs we have no real method of knowing which chemistry they use, so we would still need to fail to a safe value such as 3.3V anyway.

reversebias, 2017-07-23:
Cross posting from #29 as it's slightly more relevant here:
"Even though the caps are rated for 25V on the schematic, it's only the input filtering cap for the 3.9V switchmode converter. There's D4 (1N4148) and R30 (150R) in series with the supply to the caps, which drop about 2.5V under normal conditions. I measured 22.6V on the cap with a 25.2V input voltage. Although 25V caps are a bit marginal here, I don't think that using a 6S lipo is a safety risk."
I've been using a 6S 18650 pack for a few weeks now with no issues, the cells spend most of their time around 4V for longevity, but I don't worry about going over 24V pack voltage.

Ralim, 2017-07-23:
As I said in the other thread,
Let's keep this thread for discussions on voltage, please.

@reversebias 
I agree that 6S *should* be fine, but I need the warning in case someones iron does fail. As you are running your iron out of specifications it's up to you to decide.

reversebias, 2017-07-23:
Fair enough, back to on topic.

The battery feature is great, thanks. I found a bug where the battery voltage is not capped to the current range. Specifically the call to "OLED_DrawExtendedChar(cellV + 1, 5);", if the cell voltage is higher than the designated max voltage, the pointer offset grabs characters from the other pixel maps in Font.h. Like "ON" "SE" "TT"

To replicate: Set power source to 5S and input a voltage above 23V.

Ralim, 2017-07-23:
Ah yes, that would be a bug.
Ill fix that in the new version coming later on today. Currently working on logo support but I'll patch that at the same time.

Ralim, 2017-07-23:
The v1.15 up on github now should include a patch for this

ltwin8, 2017-07-23:
@Cpt-Huck 
3.3V per cell is okay. in FPV racing you pull like 40A-200A total on some quads even like 400A when you go very fast and powerful. 3.3V per cell would be okay because this low current (except you take a <900-1000mAh LIPO).
More complex would be the case with LIHV battery and the correct readings of them.

@EricJanVanDenBogaard meassure IR should be difficult too, because you need like a spectrum of more frequencies and you must have an current that you know. This isn't possible since all soldering-tips are a bit different in their resistance.

Just an idea: maybe for very old and little like 500mAh cells, even if voltage sags on soldering to eg. 3.2V BUT when heating element off it goes 3.8V the lipo still has energy and could used more.

Ralim, 2017-07-23:
I agree small size batteries are going to be an issue, however, if their voltage is dropping that much under a 2A load, it would make 3S nearly un-usable (and would hit the 10V cutoff). And even at the larger cell counts, if they are dropping that much their internal resistance is rather high (0.3 ohms) and they would probably barely last at all? It would be possible to only measure the battery voltage while the soldering tip is off if this becomes a large problem, however, I don't feel a massive need for this currently. (But open to ideas)

ltwin8, 2017-07-23:
Hello,

I totally agree, there is no big need for it.

LiPo's are strong enough to deliver such small current, i have an very old 4S 5000mAh and 2200mAh one, they still do. the Job.

That was only an idea for the worst case, i dont think that would need to be implemented because it is like without a sense soldering on so low spec batterys, they also only last few minutes. That was only an example from my side to show what is the "worstcase" users can apply to the Firmware.

reversebias, 2017-07-23:
I can confirm that with a small pack, the battery bar is bouncing all over the place. It drops down 2 or 3 bars between zero and full power. 

This is with a 6x18650 pack, around 2500mAh. Open circuit battery measure would obviously help here, but I may be in the minority.

ltwin8, 2017-07-23:
To my last post: 

that should just outline the worst case. Use so little lipos would be without any sense. If someone have no lipos to use take a "Turnigy 3S 2200mAh 25C" or maybe 2x 3S in series (own risk!!)

here the link, I use this as a supply for charger for my 2S lipos for my micro racers and as a flight battery for a setup that pulls.
https://hobbyking.com/en_us/turnigy-2200mah-3s-25c-lipo-pack.html?___store=en_us (3S)
https://hobbyking.com/en_us/turnigy-battery-3000mah-6s-30c-lipo-pack-xt-60.html?___store=en_us (6S)

@reversebias how about put something like few DLCs/supercapacitors at the output terminal of your pack? would cost a bit but makes the voltage very stable and I think better for the hardware too. or just use RC-LiPos because they are ratet for much (really much) higher current, that what the iron needs is almost "nothing" thats why an old RC-LiPo still does a great job...

reversebias, 2017-07-23:
@ltwin8 No real need for capacitance from my perspective. It doesn't affect operation at all, just the live battery readout. As for using 18650s, I made a pack out of what I had, just like someone would use an old lipos if they had them. Worst case at max voltage is under 3A, barely over 1C on a 2500mAh pack. Cylindrical lithum cells don't struggle at all with a 1C discharge, and I won't get puffy cells or risk dropping the iron on it.

Ralim, 2017-07-23:
I will add open-circuit measurement of the pack voltage to my todo list. Just may not happen for a bit, working on other projects most of the time at the moment.
I know where I can slot it into the firmware to (hopefully) prevent any degradation of performance.

ltwin8, 2017-07-23:
@reversebias 
have no problems until now with too puffy cells. Thats a reason more why I would use lipo, when they get puffy the worst case is fire, the 18650 can make more damage because of the packaging.

EricJanVanDenBogaard, 2017-07-23:
I have never seen a lipo pack (used for the FPV hobby) that have a datasheet, Each brand or type 18650 doo have very detailed data sheets, The lipo packs do have C ratings and max. discharge specs, but due to the nature of the TS100 these are not so important i guess.
Great job you're doing on this TS100 new firmware, Ralim !

Ralim, 2017-07-24:
Some will have data you can get by asking, but yeah normally lipo packs won't have datasheets if they are intended for racing (quads, cars etc). I mostly asked this because I have *never* seen any datasheets or official documentation that states that the cells will be damaged at such a high voltage. And if such a data sheet did exist I would seriously like to know about it since I own such a mix of cells :)

I feel that at the (relatively) low discharge rate of the iron, most batteries should cope with this fine.

closed 2017-07-31 by Ralim