Ralim/IronOS #31: Power indication on working screen

reversebias, 2017-07-23:
Enhancement suggestion for the general working screen, to include an indication of the current power usage. Maybe as an alternative to the chevrons.

After using a JBC iron in the past, I feel a seriously undervalued feature of those irons is the display that shows the current power usage (in their case, presented as a percentage). This was not only an indication of when the tip was finished heating and maintaining power, but more usefully: an indication of how well coupled your tip is to a joint or especially large ground plane.

As the tips used in the TS100 are a similar bonded thermocouple design, the reaction time should be similar. And even better, the display is right in your hand. I feel like this would be another feature that takes this iron towards professional tools that are significantly more expensive.

From reading the code, I'm not sure I understand how the heater is set, what 'gHeat_cnt' represents, or the form of the PID output values. Is an instantaneous power reading possible with the current loop structure?

Ralim, 2017-07-23:
Hmm, good suggestion @reversebias.
A power percentage could fit on the screen in the place of the chevrons if users would prefer this.

What kind of indicator would be the best to use in your mind?

Can't really do actual wattage values, as we don't have any means to measure current, and some tips are different so don't really want to have different profiles per tip style.

gHeat_cnt is the remaining time for the heater to be *on* during this *cycle*. So its kind of like a form of PWM where its a proportional on time. This is used instead of the hardware PWM timers, as its decremented by a timer tick happening, and each time it decrements the output pin is toggled. This square wave is then capacitively coupled to turn on the switching MOSFETs for the drive to the tip.
So while it's related to tip power via an odd mechanism, it is directly proportional to power, so it would be totally possible to make this be used to drive the screen with a power indicator.

The PID loop is setup to directly output in this weird scale as well.

If you want me to explain this a bit better let me know 😄 

So we can have a percentage power, but not a raw reading in watts for the display.

reversebias, 2017-07-23:
Capacitive coupling:
Ok, that makes sense, I can see C10 and C24 now. Nice to know there's a hardware solution to a firmware lock up. 

Indicator type:
The JBC uses a bargraph and a percentage, but either of these alone would serve the same purpose. As there's no small font yet (just the start screen chars), a bargraph might be easier.

gHeat_cnt:
Ok, so is the cycle time variable? And would a maxed out gHeat_cnt correspond to an (almost) 100% duty cycle?
On a high ESR battery pack, I can see the battery voltage pulsing as current is modulated to maintain power. It's around 1Hz, does this correspond to the cycle frequency, or is this PID oscillation? Ideally, the output would be stable around 1-5% power under rest, the output pulsing between 0 and 100% at 1Hz isn't as useful.

Ralim, 2017-07-23:
Yeah, it is nice to have a hardware protection against firmware issues.

I was trying to keep the font larger mostly because I have had some feedback from people who find it significantly easier than the thin font the original used. So I would agree a bar graph would be better as well.

So yeah it's basically an on time variable, the total cycle time is around 48 milliseconds per cycle. So gHeat_cnt is an "on time" out of a maximum of 48mS. So it could be basically directly correlated with a bar graph.

The 1Hz oscillation is the PID loop, mostly because I'm not perfect at tuning the PID so there is a slight tick to its response at around 1Hz. However so far because of the (relatively) large thermal mass of the tip this does not really show up at the tip temperature much so I have not pursued tuning the PID further.

So, in summary, If I made an option to display a power bar graph instead of the chevrons that would be better for you?

reversebias, 2017-07-23:
Large font is great, no issues here.

I'd like to think that a instantaneous power bargraph is intuitive, but don't want to speak for others. What is the current mechanism for the chevron selection?

That is exactly what I was going to initially suggest (the direct correlation), but then couldn't figure out from the source how it worked. I thought it was some kind of pulse length coded system. Out of curiosity, where in code is the cycle handling?

I'm not blaming you at all with the PID. So many variables of source voltage, source impedance, tip resistance, tip thermal mass, etc. Can't possibly make a perfect loop for all of the options. Maybe a slow/medium/fast response setting in the menu that changes pidSettings.ki. However this is out of the understanding of most users, and I know how you feel about extra menu items. :)

Ralim, 2017-07-23:
The current selection is just looking at the gHeat_cnt variable :smile: 
So it would not be hard to make a bar graph that matched to the gHeat_cnt variable really.

So the entire program runs on a 50mS cycle, but 2mS is chewed up in the reading tip temperature.
Basically, the code calls the two big functions in Modes.c one after another then just sleeps 50mS.
(see main.c).

I'll probably add this soon in a new update, shouldn't be too hard to make :)

Thanks, I'm glad you can understand tuning a PID. I have thought about different tunings, but just don't have the time to sit and re-tune it multiple times at the moment.

Ralim, 2017-07-27:
@reversebias 
In the latest version 1.16.x there is a power display option added.
This is probably not perfect, but is this similar to what you were expecting?

reversebias, 2017-08-03:
Nice! I like how you got the extra half level with dithering.

However, maybe I overestimated how much power the iron used for maintaining temp. For me, the power level drops down to the empty widechar for maintaining, and even normal soldering. I guess this means an idle power of below 1071, due to integer math truncation.

I think something like a logarithmic scale would be best here, to actually see the small variations down low. For the sake of simplicity though (for 0=0 and 1=1) sqrt looks pretty good too.

I propose sqrt( getIronTimer() / 30000 ) * 28. This would map 1000/30000 to 5/28. What do you think?

Ralim, 2017-08-03:
The dithering was a last minute addition to enable a full set of levels (28). :)

Yeah, for normal maintenance there is a very low power usage and it really only spikes after there is a higher thermal load. Note that I'm still attempting to tune the PID to need less input filtering (and thus faster response time) however I haven't had much success just *yet*. So the graph will become more responsive if I get that to work. 

The main microcontroller used has no floating point unit, and I'm trying hard to avoid a soft-float implementation if I can.  I might do a 28 stage lookup table for a sqrt or other correlation function to make it into a log scale display. The other option I was looking at, is to just truncate the display so that it only actually shows something like the bottom 50% since if it's greater than a certain point there is little gain in clarity. So it would just show smaller changes only?

reversebias, 2017-08-03:
I was wanting to contribute to the PID effort, as I've had plenty of experience in loop tuning in the past. I'm exclusively linux though, and still haven't got a working process for the msdos DFU. So it's a bit of a pain right now to actually iterate anything. At the moment, I update using a windows PC at work. :)

I agree with both the truncation and the lookup table. Even with dipping the tip in water, it's hard to get it close to 50% power (this is with the full 24v supply though). The resolution down low is the most important.

I just noticed that the most recent version uses 1000 full scale (line 491). Doesn't this break the chevrons in line 503? Maybe I'm misreading it.

Ralim, 2017-08-03:
I can build the firmware for you to have the PID menu at the end, that lets you temporarily change the pid settings (currently doesn't save). 

I think truncation would be easiest at the moment, I need to setup a better test system for tuning the pid and setting up the right value for the truncation.

It doesn't break the maths, as 1000 was actually a truncation already to test how it looked and I felt it looked better. 

I'm planning to look into this over this weekend, so hopefully Ill get further with setting this up :)

I also do plan to create an alternate bootloader for the system for those of us who run under non-windows environments. The only issue is they didn't include a crystal for USB, so I will probably need to figure out a method to calibrate the internal oscillator.... Or I have alot more testing to do, since I dont want to push out a bootloader that could possibly not work.

closed 2017-10-06 by Ralim