Ralim/IronOS #34: Auto Rotation bug

iLLiac4, 2017-07-27:
Hi. I have tested this feature and it works great if you are right handed. If you are left handed it rotates it to right handed too quickly.
You can try it and will see that if you solder with left hand it too quickly turns to right hand.

I hope you understand what i mean :)

Ralim, 2017-07-27:
Hi, 
I understand what you mean, I am guessing you hold your iron with the screen pointing further up than I normally do. I will push a bug fix version shortly that should increase the threshold required. :)

iLLiac4, 2017-07-27:
Yes i am holding it as pencil.

Ralim, 2017-07-27:
Original there was only a minimal threshold, so when the screen rolled past flat the screen would flip to the other orientation. I am bumping up the threshold so that it will not rotate until the screen rolls further past flat. Just testing the build on my units and then I'll push up a patch.

Ralim, 2017-07-27:
@iLLiac4 Can you try the latest v1.16.1?

iLLiac4, 2017-07-27:
Much better. Probably this is hard to do but when you pick them up from the table and if it flips to right hand display you have to make it almost vertical to flip it to left menu.
But as i said this is probably hard to achive.

Ralim, 2017-07-28:
Hi, 
I'm not quite sure what you mean by this :(
Which axis are you rotating around? 

I will be polishing this a bit more in the coming days, as I think I have found an edge case I need to test soon.

Ralim, 2017-07-31:
I'm going to close this for now, please open this again if the issue is not resolved.

closed 2017-07-31 by Ralim