Ralim/IronOS #35: Value changing in settings should be faster

Cpt-Huck, 2017-07-30:
when in settings and changing values.. I can press a button to go upwards..
 and while holding the button, it goes automatically up..
for for some settings like shut down time, where the range goes from 1 to 60 min ?!
long pressing is not fast enought..
it should be twice as fast...
so you can change values even faster.
understand what I mean ?

Ralim, 2017-07-30:
Hi,
I know the issue you are talking about, I was actually already working on this in my spare time :)
I just pushed up the faster auto-repeat version to the releases page.

Let me know what you think of this? :)

Trezona, 2017-07-31:
The latest release is great, although I feel the auto-repeat could be a little faster.

Ralim, 2017-07-31:
Fair enough, ill bump it up a bit more in the next release :)

Ralim, 2017-08-03:
@Trezona It's doubled in the latest release, thoughts?
Cant really push it up much faster than this at the moment. :)

Trezona, 2017-08-03:
Just tested now, much better for me. Thanks.

closed 2017-08-04 by Ralim