Ralim/IronOS #36: Few suggestions.

Gregory1981, 2017-07-30:
Hi. 
Congratulations for your current job.
Well done. :)
Anyway I have few suggestions.
1. Temperature degree symbol just gone in furhter release. Is possible to took it back?
2. During normal operation temperature font could be bigger like in original firmware. In my opinion looks better when is fill whole resolution of screen.
3. Temperature change style looks better on original firmware (flipping) than just static value change.
4. After turn on power supply screen sets in automatic mode for left hand side. It is quite annoying. Is that possible to check by acceloremeter firstible position and then set the position of screen?

Of course I can understand you may not take into account my suggestions, but will be nice if you will consider them in further releases.

Kind regards,
Greg.

Ralim, 2017-07-31:
Hi, 
I certainly will consider them all :) 
To answer you questions:

1. It's coming back don't worry, just need to redraw it slightly to make it fit in the re-design. It's on my list for the next update. 

2. Do you mean the current tip temperature when you are soldering? I'll have to check but I thought off the top of my head the font was full height. 

3. I plan to add back in a movement when you change the temperature on the screen, it's just a bit of messing around that I haven't yet had time for. 

4. At the moment at start up it does read the current orientation, it's just the imu defaults to left handed side before it initialises. I will probably flip this around so it will default to right handed in the next firmware though, since the majority of users are right handed. 

5. Absolutely, I only used 100 to reduce the pressing the button before I had added the auto repeat. I'll definitely be changing that in the next firmware. 

6.its all built in the system workbench for stm32 software. As I'm only using open source compilers and tools. (good luck getting the paid ones running under Linux natively). 

If you could just clarify what you would like changed with 2 it would be awesome :)

Gregory1981, 2017-08-01:
Hi. 
Thank you very much for taking my suggestions under consideration. I really appreciate that.
About point number 2.
OLED display resolution is 96x16 (I guess). In original firmware, temperature font fills display fully, so height of font is 16 pixels. And at 1.16.2 release font height looks like 10 or 12 pixels. Maybe I am wrong. If yes, please correct me. If not, will be that possible to change to 16 pixels?
Thank you in advance.
Kind regards,
Greg.

Ralim, 2017-08-01:
Hi,
No problem at all, it's what I like to do :) Means the Iron gets better for the entire community.

Yes, you are correct on both points, the display is 96x16 pixels, the font is fixed width at 12 pixels wide. So this results in the digits being around 10-12 pixels high as that is the font I used for the text when I created a bolder fontface. I have been toying with creating a new taller font, however, most of the fonts I have tried so far have ended up looking a little too stretched, where they looked taller than they should be. I will have another look at this though, but I'm trying not to widen the font anymore than it already is. I just need to find a good looking number set that is okay in a 16x12 pixel format.

So its completely doable, I just havent put down more time to create a new font for the numbers yet :)

Ralim, 2017-08-08:
@Gregory1981 
Just a small note to let you know that the newest release alpha I've put up and all following firmware has a new font, that is a better fit and looks a lot better in my mind.

I'm still planning to bring back the temperature symbols too, along with more animations in the future.

Gregory1981, 2017-08-08:
Hi.
Thank you very much for new update. New font looks amazing. :)
Blinking during cooling down to safe temperature is great feature. Thanks a lot for that.
Thank you for extend the standby temperature to 50 degrees.
Screen orientation nicely set now itself to right hand orientation after power up, but after one second flip back to left hand and then again back to right hand :) Small hiccup? Maybe acceleration sensor issue in my device?
Is there any way to donate to you for your work? According to fact you doing it voluntary I think you deserve for donation and support to motivate you or just to say thank you.
Kind regards,
Greg.

Ralim, 2017-08-08:
Glad you like the new font :) 

I'm still trying to figure out the best way to handle the orientation detection. Because the accelerometer used in the iron defaults to the left handed mode, there isn't really a huge amount I can seem to do. I. Still trying other ideas, but at the moment I tend to just hold it in the right hand mode as it powers up and that seems to help it stabilize. 

Because of the way it works with the thresholds, if you don't tilt it enough "right handed direction" it doesn't latch that orientation and stays the default. At the moment if you always use the iron in one hand, one option is to just set it to be right handed in the settings. 

I doubt it's a problem with your device, so far all the units I've used have been built quite well.

I will let you know if I get further on the accelerometer though, I do still toy with it (increased its reliability in this update, but didn't fix this bug). 

If you want to donate there is a link right at the bottom of the main page for a PayPal donation, but if that isn't convenient your always welcome to email me for other methods (PayPal is just easy)

Gregory1981, 2017-08-08:
Thank you for answer. 
Tryed to find the link to paypal but can't. Is your email linked to your PayPal account? 
And just found another issue. When screen is orientated upwards or is orientated to the left device is not going to standby mode. Maybe way how acceloremeter check the movement doesn't react to one of the axis or when one axis (Y) is orientated close to its minimum or maximum value and reads only values on the middle of the range not close to the minimum or the maximum? Just found now it goes to sleep when screen is around 45 degrees of Y axis (between upwards and to one of the side). Im not sure if I explained that properly.

Ralim, 2017-08-08:
Hi, 
https://paypal.me/Ralimtek is the link, but it is also attached to my email afaik. 

Hmm that's an odd bug, that shouldn't be happening. The settings for axis are the same (I'm using the accelerometers built in movement detection rather than checking the values myself). I might suggest maybe turning down the sensitivity and testing if you can? I'll try and test it further tomorrow as well. 

I think I get what you mean :) and I'll try and play to reproduce it tomorrow. If I can reproduce it then there is a good chance of me being able to fix it :P

Gregory1981, 2017-08-09:
Release after release is much, much beter :)
I was quite sceptic about sleeping time in seconds, but when tryed it I really foud it useful. I dont't use soldering iron at home to often and was alwas worry about burn myself or burn surroundings. But when tryed the newest firmware I really found very useful. Thanks a lot for newest release :d 
And now found there is no problem anymore with getting device to sleep in any position, thanks a lot for that :)
Do you think add blinking feature during sleep mode will be ok?
Kind regards,
Greg.

Ralim, 2017-08-10:
Glad to hear it's getting better for you. Users enjoying the features is the reason I like to keep adding features (Plus I get to enjoy the features too).

The blinking during sleep was an accident. Its only meant to be blinking in the cool down mode (and only if its turned on!). That has hopefully been fully patched in the latest release :D

closed 2017-10-06 by Ralim