Ralim/IronOS #37: Feature Request: Short click on the boost button should change temperature

awigen, 2017-08-02:
I think it would work quite well if the boost button also worked as a temperature select button.

If the boost button is pressed for less than 200ms go into the temperature menu.
If the boost button is pressed for 200ms or more do normal boost mode.

Ralim, 2017-08-03:
At the moment the button code does not latch down times and instead looks at the total time down so far.
So at the moment this is not particularly easy to implement, which is why it's not currently there 😄 
I agree completely this would be the most logical next step to add, however.
I'll have a better look into the button handler in the next few days and get back to you on this.
Hopefully it shouldn't be too hard to add :)

Ralim, 2017-09-26:
This has been added to the 2.x branch as a new feature (with the different button handling).
I'm closing this issue off as it will be in the 2.x branch that's coming soon (when I nail a bit more).

closed 2017-09-26 by Ralim