Ralim/IronOS #38: Possible translation offer to es_ES.

Aglezabad, 2017-08-03:
Good evening:

Recently I've acquired a TS-100 device and I like your firmware. I'm interested in translating your software to es_ES (Spanish from Spain).

The question about this is: How to do that? Maybe at compilation time with preprocessor checks? The thing is how to avoid the extra effort in the future.

I will wait your answer before doing anything.

Best regards.

Ralim, 2017-08-03:
I was thinking of performing the language at compile time.
I'm in the process of splitting all the strings out to a strings.h header file for all the text.
Would es_ES require extra symbols in the font map? (If so, no problem, I just need to make more).

If you're interested, do you think it would be possible to translate the text keeping the same limits on names in the settings page? (Since screen width is fixed at 8 chars, settings names are really short).

Could at the first pass just translate all the help text and see how that goes :)

So from here, I'll hurry up and move all the text out to a separate file, so you can just provide a separate translated version of that file. And then when I build the firmware, I'll do a build for each language :)

Aglezabad, 2017-08-04:
Perfect. I will do the translation with the most recent commit, so I can tell you which changes are needed.

Ralim, 2017-08-04:
Yeah, I'm currently working on moving all the text out to its own file. But if you can even initially do a rough pass for the text you normally see on the iron, and also any extra symbols you would require :)

Aglezabad, 2017-08-05:
Hi:
I've done a rough translation for testing purposes, only editing the text in Modes.c. For this moment, I don't need any extra characters because I've omitted the accentuation needed in my language (á,é,í,ó,ú). So, it's not neccessary to modify the font.

Ralim, 2017-08-05:
Awesome, thank you :)
Not modifying the font makes it easier, though if the accent's fit I'm not opposed to adding them in future :)
I'll make a test build probably tomorrow and upload for you to test :)

Ralim, 2017-08-06:
Hi,
Can you please test/check the attached firmware.
I used your translated strings to build the firmware. However I currently haven't coded quotation marks (" ") into the font yet, so I've replaced these with the <> for now.

Let me know what you think :)

[ts100.zip](https://github.com/Ralim/ts100/files/1203005/ts100.zip)

Aglezabad, 2017-08-06:
Hi Ralim:

Thank you for your build. I will check today. Also, I'm a Ubuntu user, so I will test the flashing script for Linux too.

Regards.

Aglezabad, 2017-08-06:
Also, I've changed the title, because the translation is going to be a reality.

Ralim, 2017-08-06:
Awesome, let me know how both go 😃 
Yep, I think the translation is a good idea :)
If it works out okay, I actually pushed up another option. If you could translate the following too it would be awesome :)

"      Automatically starts the iron into soldering on power up." 

Just answer the translation here, no need to upload the file for one line :D

Aglezabad, 2017-08-06:
The string translated: " Iniciar directamente modo soldadura en el encendido". It's not a literal translation but it has the same meaning.

About flashing in Linux, it does not work for me. This is my environment:

Ubuntu MATE 17.04 64bit
Kernel 4.12 (Custom, compiled by me)

I will check a solution for that (in another issue). But, to avoid delays on testing, I will flash in Windows 10.

Aglezabad, 2017-08-07:
I've flashed the translated firmware and it's working well. I want to know where are the other strings that I must translate to complete all translation.

Ralim, 2017-08-08:
Hi,
I believe I have moved all the strings into [strings.c](https://github.com/Ralim/ts100/blob/master/workspace/ts100/src/Strings.c)

If you could translate all the options in there as you feel appropriate?

At the moment the text "ON" / "SETTINGS" that appears on the home screen cannot be translated as they are burned into the image. When I rewrite the graphics driver for these they will be added to the strings.c file to allow for translation.

Ralim, 2017-08-08:
I have pushed out an alpha of 1.17. This should be complete pending a new setting being translated and your testing. :)

Also, the new font does add the rest of the common ascii set. Not sure how it works out but I'm looking forward to being able to support accentuation so it looks better for you :)

Aglezabad, 2017-08-09:
Hi Ralim:

Thank you for your work. I haven't checked the new alpha version yet. I will try to test today if possible and give you the rest of translated strings.

Regards.

Ralim, 2017-08-09:
Just an update to this. The translation worked well, except my fonts not in unicode order so I need to re-order some of the glyphs for it to work. So it will be coming soon :) Just need to spend a bit of time shuffling stuff around :D

Ralim, 2017-08-11:
@Aglezabad Could you please test the release I just pushed up. Should add support for the accentuation. Let me know you spot any errors (you will spot them way faster than me).

Aglezabad, 2017-08-11:
Ok, i will try it asap.

Aglezabad, 2017-08-12:
Hi:
I have updated the ts100 and I really like the new font. Good job. I didn't found any problem nor typo in the screen, so it's working for me.

Also, the short words of the settings menu will be the same, to avoid confusion. So the translation is done for now.

Have a nice weekend.

Ralim, 2017-08-13:
Awesome, I'll close this issue for now, and Ill re-open it & tag you if newer firmwares need translation :)

closed 2017-08-13 by Ralim