Ralim/IronOS #39: Feature request: Cooling down temperature display should blink until it reaches "safe" temperature

ErhardDu, 2017-08-04:
Would be nice to have the temperature display blinking until it's down to let's say 50C to signal that the tip is not safe to touch.

Great job!

Ralim, 2017-08-06:
That would be a good idea.
I'm thinking maybe invert all the colours on the screen for the blinking. What do you think?

ErhardDu, 2017-08-06:
Great idea! So it's not necessary to look at the display directly and you can notice it from a different angle of view...
Thanks for your quick answer...

Trezona, 2017-08-06:
Ralim, will this be a switchable feature in the menu?
I still prefer seeing the actual temperature dropping.

Ralim, 2017-08-07:
@Trezona  Of course! Anything where people could possibly want different ideas I add it as a menu option :)
I was planning to see the temperature there still, and just have the screen invert colours while cooling.

Ralim, 2017-08-08:
I have pushed out an alpha of 1.17 that adds this as just blinking the temperature. I'm not sure if ill do full-screen inversion or not, what do you think?

Ralim, 2017-08-11:
@ErhardDu 
Have you tried the latest releases to look at this? Is the blinking okay or would you prefer full inversion?

ErhardDu, 2017-08-11:
Awesome!
For me it would work best if the whole thing  would invert.
Thanks for your effort!

Ralim, 2017-08-13:
@ErhardDu The latest (1.17.5) release adds display inversion instead. Can you test this please ?

csdpm, 2017-08-13:
I know I'm not the original thread owner but to me, the full screen flashing inversion is, um, horrible (sorry, just one opinion, everyone else might really like it!).

P.S. @Ralim: Can you please re-open #62

Ralim, 2017-08-13:
I personally don't like inversion, but as it's a feature that defaults off* I just leave it turned off.

I'll wait and see what others think, I can easily change it back to just blinking the temp if that is what people prefer.

* I only bump settings version on full releases, which means if you follow along on alphas not all settings are 'reset' after programming.

RE: PS normally just wait and I'll re-open the issue. I'm not always near my email to respond rapidly :)

csdpm, 2017-08-13:
Many thanks Ralim - I wasn't trying to hassle you, I'm just not that familiar with how github works yet when it comes to closed issues, ie. whether you would automatically see it. :)

Ralim, 2017-08-13:
I get an email for *any* reply to *any* issue atm :) Or you can use the @ symbol to tag others into the conversation.

csdpm, 2017-08-13:
I should have guessed that! Just me being dumb, I should have realized. Thanks for the tip.

Trezona, 2017-08-14:
I think i prefer just the temperature flashing!

csdpm, 2017-08-14:
> Edit: Would it be possible to have "Cooling" displayed until 50 degrees then display "Off"?

It just goes to show how many different viewpoints there can be on one display aspect.... If anything, I might have said the opposite and suggested 'Off' when it turns off and 'Cool' when it is cool (It might be different if there was space to write 'Cooling')  :)

Trezona, 2017-08-14:
Yes i was hoping that the text size could be reduced to fit in "Cooling" during the cooling fase, then display "Off" when it reaches 50 degrees, which is actually in an "Off" state :-)

EricJanVanDenBogaard, 2017-08-14:
would also be nice > when cutting power, and applying power again, when tip is still hot, show flashing hot temperature again until cool down and go to iron icon again, or other idle mode

Ralim, 2017-08-14:
I think it is certainly possible to have the text change once the unit has cooled down. At this point in time, all text is created from a pre-rendered font that is a fixed size. Could I add scrolling text instead? Would this be more or less annoying?

@EricJanVanDenBogaard It would be easy to have it check the tip temperature, and if its hot have it go automatically to the temperature warning screen (aka cooling screen) ?

EricJanVanDenBogaard, 2017-08-15:
That would be great !

Ralim, 2017-08-26:
This should be closeable with the latest 1.17 release.
The temperature blinks (not the inverting screen).
The iron will also jump to the cooling screen with temperature if it starts up and the tip is still hot.

closed 2017-08-26 by Ralim