Ralim/IronOS #41: TS100 Logo Editor under Linux: howto

doegox, 2017-08-05:
Hi,

FYI I could compile and use the  TS100 Logo Editor under Linux, but this requires setting .Net dependency to v4.5:

```bash
cd Logo\ GUI/TS100\ Logo\ Editor/TS100\ Logo\ Editor/
sed -i 's#<TargetFrameworkVersion>v4.0</TargetFrameworkVersion>#<TargetFrameworkVersion>v4.5</TargetFrameworkVersion>#' TS100\ Logo\ Editor.csproj 
 xbuild TS100\ Logo\ Editor.csproj 
```

Ralim, 2017-08-06:
Interesting, my local install of mono would only run it nicely under 4.0 :/

doegox, 2017-08-06:
There is no page on the wiki, all the procedure for the logo is currently on the main readme so I propose you include the notes for Linux there too.
Here is a little snippet to be added to the "Setting a custom logo" (or move all to the wiki but I didn't feel like adding a new wiki page just for these 4 lines):

Note for Linux users: the GUI can be launched directly with mono (`mono TS100.Logo.Editor.exe`). You can also compile it from the sources with `xbuild "TS100 Logo Editor.csproj"`. Depending on your Mono version, if you get `The predefined type ... is not defined or imported` errors, you will have to change `TargetFrameworkVersion` in `TS100 Logo Editor.csproj` to match yours, e.g. from 4.0 to 4.5.

tigert, 2017-08-07:
Seems that Wine runs the provided Windows EXE just fine.

However, since this is just a simple file conversion where you really want to draw your image to the correct pixel size, wouldn't a simple command line script or tool work best?

Ralim, 2017-08-08:
@doegox Fair enough, Ill probably be splitting the entire custom logo out into a Wiki page soon, I'll add this in there.

@tigert Good to know wine runs it well. I'm sure someone else can do it in a command line script, but I use C# for desktop apps normally during work so its what I have ready access to. More than happy for someone to create an alternate version for other operating systems if they desire.

tigert, 2017-08-08:
Just for the record, mono opened the binary EXE fine too.

   ```mono TS100.Logo.Editor.exe```