Ralim/IronOS #42: Issue: after writing 1.16.3 successfully (rdy) and powering up old original fw is running...

ErhardDu, 2017-08-06:
Hi,
copying to ts100 is working, disconnection/reconnection also but after rebooting with 24 V my original fw is showing up. Tried two times, both the same...
regards Erhard

Ralim, 2017-08-06:
Hi, 
Can you confirm what system you are using to perform the upgrade, what the file extension changes to after you copy the file?

This is an issue with the built in DFU software, which I cannot fix. But more than happy to try and get this working for you :)

ErhardDu, 2017-08-06:
Hi, I copy from Win10.
Currently there is **fw ver 2.12** on my ts100 and **DFU:3.43**.
After copying ts100.**hex** changes to ts100.**rdy** as mentioned in your howto.

Thanks for your efforts!

Ralim, 2017-08-07:
If the file changes to .RDY after you copy the file it should have flashed the file correct. Everyone's irons are currently running DFU 3.43 as that's the only DFU we have for the iron currently. 

Honestly, If it's changing to .RDY it should have flashed the new firmware properly. Can you possibly try a different computer? Their DFU bootloader seems to have issues with some computers for no apparent reason.

Ralim, 2017-08-11:
@ErhardDu Have you made any further progress on this? Can you confirm if you are using the windows built in copy feature or another tool?

ErhardDu, 2017-08-11:
Hi, on my second try the whole thing works for me. Must have made a fault the first time.  Great!

ErhardDu, 2017-08-11:
By the way: What does the F and T in the settings mean? Is it an abbreviation?

closed 2017-08-11 by ErhardDu

Ralim, 2017-08-11:
F / T = False / True
Glad it worked on the second try. No idea why it didnt the first time around.