Ralim/IronOS #46: Smaller delay for sleep mode ?

Vidalcris, 2017-08-08:
Hi,
This firmware is awesome but i just would like to be able to set the sleep time to 0.1min.
Do u think you could add this in next update ? :p

Ralim, 2017-08-08:
Yeah I'm sure I can add something. 
I'm not sure how best to handle it though. Currently my best idea was to add a setting for fast mode, that divides it by 10. So it would change it from 1-30 minutes to 0.1-3 minutes. 

Thoughts?

Vidalcris, 2017-08-08:
I was thinking just using seconds instead of minuts..
I personnaly dont see the point of using a sleep delay longer than 1 minuts ;)
You could set the max value to 600 then.. and minimum to 1 !

Ralim, 2017-08-08:
Alot of people use it set to around 5 minutes, as they don't want to be waiting for the heatup cycle everyone to come to use it. (ie if your making up cables, so you solder a joint then heatsink etc and come back to it cool again). Also don't want to put too many heat cycles on the tip (though not actually sure if that's better or worse than staying hot). 

I'm not that cruel as to make someone do that loop if they go past their setting. I might try and make it roll through 10,20,30,40,50 seconds and then go into minutes up to 10 as you change it.

Vidalcris, 2017-08-08:
Oh ok i see :)
I'm using it for small things when i'm working on my quadcopter (i mean during 1 hour of work i will use the iron around 1min : 6 x 10seconds)
Thanks a lot in advance for this modification :)

Ralim, 2017-08-09:
The latest release should have added this :)

Vidalcris, 2017-08-09:
Thanks a lot ! you are the best ;)

closed 2017-08-10 by Ralim