Ralim/IronOS #48: Voltage cutoff when not soldering?

tigert, 2017-08-08:
I tried the new firmware today with a (non-full) 4S LiPo, it correctly showed an almost empty battery icon.

I was testing the "blink temperature until safely cooled" feature, and then forgot about it, leaving the iron plugged on the lipo on the table. I was later surprised to find the battery at ~1.5V per cell, so something had drained it.

Would anyone have an idea how much power the iron consumes when it is not heating? And, more importantly, does the low voltage cutoff system function also when the iron is not in soldering mode

I will investigate this further, but just wanted to let you know in case someone has time to poke at this.

(Since the battery was probably drained with low current, I think it'll be OK when charged with NiCd program at 0.1A for a while)

Ralim, 2017-08-08:
I don't have the measurement near me but it does pull a few ma when running. Low voltage does not run when not in soldering mode, otherwise it can become a pain if you set it incorrectly. Ie if you set it to 5s then use a 3s it would turn off as it's too low. So the tradeoff at the moment is that the cut off will stop the iron heating and running the battery quickly but can't really stop the iron pulling its quiescent draw. 

That said there is a fix coming for the screen not sleeping on the cool display.

tigert, 2017-08-09:
Yeah, true. 

I guess if voltage at boot is lower than the set cutoff voltage, one could have a timeout for the power-off so you can change the setting? 

Or I wonder if you can reliably detect if you are on battery vs power supply (voltage drops over time) and do some logic based on that.

Ralim, 2017-08-09:
Yeah, fear of locking someone out has stopped me trying anything particularly complex.
Also, the unit only draws 0.2W when running (with the screen on), but with the heater off. So most batteries this really doesn't do much very fast. Even with the screen off the unit draws 0.18W.

The iron cannot turn off its internal power supply, it can only turn off the screen. So there isn't a huge amount really to gain here either. The latest firmware did bring back the screen going to sleep again which will help a *tiny* amount.

There was a discussion on this a while back, and basically, we have very little ability to tell battery vs power supply. Especially as people use cheap supplies that drop over time as they heat up (or age). And then there are people like me that use a 6S@10Ah battery most of the time, which has less drop than some power supplies :P

GS88, 2017-12-06:
I use salvaged 18650 cell from laptop packs. Most of them have higher internal resistance but lots of usable life left. It would be very very useful to have a lower cut out for these cells. 

I usually cycle my salvaged 18650 down to 3.0V on loads <1C. 

Could you add a setting to adjust the per cell cut off? that way people using expensive LiPo's can keep it higher, and people using junk 18650 can keep it low and have a longer runtime?

A range of 3.0V to 3.5V would cover 90% of lithium-ion use cases.

Ralim, 2017-12-12:
I could probably just drop the cutoff down to 3V/cell in general, as afaik li-poly is ok with that as a lower cutoff too ?

GS88, 2017-12-12:
All lithium cells are ok down to 3.0V most are fine down to 2.5V occasionally at higher discharge rates. The problem comes when you have unbalanced cells with no protection circuitry.

In unbalanced packs, the lower capacity cell will sag much lower than the rest of the pack and become damage making the balance problem worse.

I can understand why RC hobbyist baby their Li-Po if they are unprotected and want to keep a very low internal resistance, but MOST low current draw cases you want a cutout at 3.0 to get the most capacity out of your packs. 

https://cdn.sparkfun.com/assets/6/7/4/7/9/5112a224ce395fb479000003.png
Sparkfun link above shows a 2Ah lipo discharge curve at 6C (12A).

https://tinyurl.com/y9rocxm9
The link above is a picture of lithium discharge curve at different rates.

3.0V is almost always the best cut off voltage if you have a protected pack.

tigert, 2017-12-12:
Yeah, I guess we have two cases: LiPo flight (etc) packs, most common 3 and 4S, some 5-6S. And the 18650 LiIon cells, that also have the same cell count choices I guess. Maybe have the menu be:

Power Source: AC / LiPo / 18650 LiIon
IF battery source, then enable cell count menu: 3-6S(?)

Then set the cutoff voltage based on the power source cell type?

tigert, 2017-12-12:
Reason for above being that LiIon does have a lot lower low limit as far as I know. I am pretty sure people do want to save their LiPo packs, as they are indeed unprotected - although most salvaged laptop 18650's are too.

GS88, 2017-12-12:
Virtually All li-ion and li-po cells will last their rated cycle life or longer at 3V to 4.2V. 

Some manufacturers specify more aggressive voltage limits but they will only be higher charging voltage or lower cutouts. In those cases, a cell will operate at lower than rated capacity but have a longer cycle life at the more conservative charging and cutoff voltage.

Cells generally age with the number of cycles by losing capacity and gaining internal resistance. less aggressive voltage limits can significantly improve the cycle life of any battery.

Internal resistance degradation is the reason why hobby lo-po's are treated any differently from li-ion or even li-po in a laptop. Hobby packs are rated for peak amp performance and aging effects can reduce that performance. 

The individual cells in Hobby lipo packs will be made by different companies than the ones that package and brands them. The pack may have drastically different specs from the cell manufactures. 

Anyways hope I am not droning on.  I really like the work that's been done on the TS100 firmware. Thanks for all the work Ralim and everyone have done. 

I use a couple TS100 irons to build battery packs and protection circuits all the time. I have 5s and 6s packs powering my Irons.

Ralim, 2017-12-13:
I initially went for a cutoff of 3.3V mostly to play it safe, since a lot of the quad fliers I have found have little to no balancing/protection on their packs.

Do we really want to add another menu option :grimacing:  or do we drop it to 3V per cell ?

ajanulis, 2018-04-03:
Sorry if it was discussed earlier - I tried to find out but got no answer why such a nice and universal soldering iron is treated as a tool for RC hobbyists only? 3S - 6S or DC. I might be wrong but TS-100 is almost perfect for the majority of electronic enthusiasts. And this means that we (DIY) might be using way more wider spectrum of batteries (like LiFePo4, AA, NiMH or even some of the mobile screwdrivers). I understand your wish to have as minimal settings as possible but having just a regular cutoff voltage (in 0.1V steps) should please all the needs (and I truly believe that RC guys are really smart enough to choose the right voltage). Am I wrong or I miss something essential?

JohnEdwa, 2018-04-03:
Mostly because only hobby LiPo batteries need a cutoff voltage protection on the device - if you use something else, it either doesn't matter (AA & NiMH), or you really should have a protection circuit on the battery itself.    
Sure, adding 0.1-volt steps is possible but with the range being from 9V up to 24V and you only have one button, have fun waiting for the value to scroll 120 times to change from 5S to 4S. Spoiler alert: you won't, and end up ruining your lipo.

Ralim, 2018-04-04:
@ajanulis , @JohnEdwa is on the point perfectly here. Also keeping in mind that you need a source that can supply ~2A, so AA batteries wont particularly like this. Also LiPo's are just ridiculously common these days, and most people using other batteries have their own protection or know what they are doing.

ajanulis, 2018-04-04:
OK, thanks @Ralim, @JohnEdwa I see - you prefer speaking and arguing instead of doing. I agree, my fast example with AA wasn't good, but I'd really like to have an option to use Dewalt 18V battery for soldering on the field. And, by the way, have you ever tried to set some electronic clock, just holding some buttons? Was it overcomplicated? Even stock firmware has this option - sadly in a short range. It always drives my crazy when I should add some additional HW just because someone is just lazy to add few more lines to code - it doesn't hurt anyone and is a way more sustainable way to our planet. Sorry if it sounds a bit rude, but it's how I feel about it. Peace.

Ralim, 2018-04-04:
No, it's not about arguing.
This is a discussion, and it's worth having a discussion if you think this is worth a change. The firmware originally had a voltage select but I received emails asking for it to be easier which is why we have what we have now. 

Most 18V drill batteries are lithium? Shocking I know, the 18V drill batteries I use are 5S lithium batteries.

I agree that it's easy to have the buttons change menu options. But do most users who are swapping between 3/4/5S batteries for their quads want to have to cycle around every single time they pick up the unit?... Not particularly. I agree yes it's not a big hassle, but if in one day you have to change it more than once it posses you off.

One option here would be to add another menu entry for *user* and then let you adjust that in a submenu.

It's just more to add that often isn't needed and that needs to be kept around for future.

Keep in mind that most of us don't have much time to work on projects for free, and if you want us to add a feature the best way is to be nice :D I'm not offended man, just a heads up for your future projects.

ajanulis, 2018-04-04:
Yes, in this particular case, Dewalt 18V is 5S. And not, Dewalt 20V is not 6S. @Ralim, I am just wondering why there is a tendency to limit your firmware "customers" just to RC hobbyists? Is it cultural? What country are you from? It somehow reminds me Apple and it's persistence to add such useless (for a lot of people) apps like Stock, Games etc) and, in other hand, very limit possibilities to fine tune your own phone as you wish (and sadly looks like they succeeded - at least no jailbreak for a couple of years). I agree, that for certain group of users less is more and it might be annoying to have a lot of unnecessary functions and options you can't skip, but maybe having Advance section can be a good balance between regular RC guys and others?

JohnEdwa, 2018-04-04:
@Ajanulis 
>  And not, Dewalt 20V is not 6S.

Yeah, [they are misleading marketing wank.](http://toolguyd.com/dewalt-20v-max-18v/) and exactly the same 5S Li-ion batteries but they market the maximum voltage instead of the nominal voltage.

Ralim, 2018-04-04:
It's not cultural, it's catering to the masses. 
The vast number of users use a lip battery. 

Also since the cut out voltages are roughly (10V, 13.3V, 16.5V, 20V) these also line up roughly with lifepo4, where they will cut off the battery slightly early but not before the vast majority of the cells capacity has been used.

For protected cells most people use the DC mode. 
For AA,  NiMh and NiCad you are going to be not as concerned with the cut off, and a rough value usually works.

So I'm not trying to fight with you here, I'm just not sure of the use case that requires other options? Because that's all of the common chemistries?

Ralim, 2018-04-04:
Also, the reason why it is set high at 3.3V per cell, is because I could not find a single datasheet for a lip that suggested a higher cut off after going through all of the lithium batteries I have designed into products, which is why this very concervative value is used. While it means you don't quite get the full capacity from some cells, it usually works out to 90%+.

ajanulis, 2018-04-04:
OK, convinced and reflashing :) And no, @GS88, not going to support this consume consume rat race just because of a lack of some tiny code for fine tuning :)

GS88, 2018-04-04:
most lithium power tool batteries actually have cut out voltages at 2.5V.

https://akkuplus.de/mediafiles/Datenblatt/Panasonic/Panasonic_NCR20700A.pdf, that is the battery in the new Dewalt packs. almost all Li-ion 18650 cell can handle 2.5V cutoffs with minimal degradation.

The 4S setting on the iron would give you a cut off at 2.64V using a 5S pack... problem solved!

ajanulis, 2018-04-04:
@GS88, nop: 

https://syonyk.blogspot.no/2016/09/dewalt-20v-max-30ah-battery-pack.html

JohnEdwa, 2018-04-04:
@ajanulis, yup:

https://syonyk.blogspot.fi/2017/04/dewalt-20v-max-60ah-pack-teardown.html

Literally from the same blog.

GS88, 2018-04-04:
Syonyk's is one of my favourite blogs!

BTW the new 20700 cells in the Dewalt packs are amazing. They come from the same pedigree (Panasonic) as the cells Tesla makes at the Gigafactory. 

Dewalt have the best batteries of any tool manufacturer right now. Their pack design does have some trade-offs though...

those if it does not have active management built-into the pack, it relies on the charger and tools for balancing and protection.

Ryobi batteries altho using cheaper cells have a good management system with protection built right into the pack. They make a much better battery to run your soldering iron from in DC mode.

GS88, 2018-04-04:
I doubt DeWalt fulfil many warranties on their batteries. Old "dead" tool packs are a great place to get free high drain Li-ion cells for projects that need them.

Actual relatively few companies put a full low voltage cut off in the pack, most just have the sensing electronics that tell the tool to stop working. Rigid is another one I know for sure has a cut off built right into the battery.

Tool batteries are design to do 15A - 30A continuous with higher peaks, that requies a mosfet with cooling and adds cost. The majority of tools use a mosfet with some cooling for speed controll any ways, so it cheaper just to out put a no go signal from the battery to the tool controller.

Syonyk, 2018-04-05:
Syonyk here.

Despite the datasheets for lithium batteries going down to 2.5V, that's the sort of thing you do when you're a manufacturer testing to show the absolute maximum possible capacity.  It's not the sort of thing you want to do in the real world to cells you actually like.  Any of them.

Based on fairly extensive experience with lithium ion cells, I consider 3.0V a very reasonable low voltage cutoff for any sort of 4.2-4.4V chemistry (so anything "not weird").  That would give a 4S cutoff voltage of 12V, a 5S cutoff voltage of 15V, and a 6S cutoff voltage of 18V.

Using 2.5V gets you an extra 200mAh or so (on the 20700s - less on 18650s) at the cost of really stressing the cell and hurting cycle life a lot.  Lithium is most fully stressed at fully charged and heavily discharged.

Also, the pixels being spilled about the differences between hobby lipo packs and liion 18650s are quite missing the point - the chemistries are the same, only the packaging is different.  Lipo simply refers to the soft pouch cells, and says nothing of the chemistry (though typically a hobby pack is LiCoO2, since it's a really high energy/power chemistry - you pay for that in poor cycle life and a slightly scarily low runaway temperature).  You can get cylindrical cells in the same chemistry if you want, or plenty of other options.  But, despite all that, you still probably don't want to be running any of them to 2.5V if you can help it.  So the 3.0V/S failsafe is a good point to use.

I understand the point about not wanting to lock users out if they change cell capacity, so a reasonable option would be to not enable low voltage cutoff until soldering is enabled - and then have some sort of failsafe at around 12V.  For a 4S pack, that's 3.0V/cell (empty), and for a 6S pack, that's 2.0V/cell (really, really empty, but not low enough that it's particularly dangerous if you recharge quickly from there - it's lower than I like to see cells, but I wouldn't worry about recharging one unless it had been sitting there for ages).

Anyway, neat project!  I have no particular use for it (I use a 150W butane iron for portable work), but I'm glad to see the hacking of commercial products to make them better!

JohnEdwa, 2018-04-05:
@Syonyk, I totally agree on not letting them drain to 2.5v repeatedly, that is their 'minimum safe vvoltage' an in no way healthy for them in the long run. On most cells though, that value is usually 3.0v, so these *should* be able to handle a little more. Still, no reason to push them just because you theoreticallly can. 
One should always remember that the protection circuits aren't there to actually protect the battery, they are there to protect the user and prevent fires and explosions. If you get a low voltage cutoff, you went too far (most BMS's that's around the 2.5-2.6v mark).

In the quadcopter world, the rule of thumb is that a cell should recover to 3.6-3.7v, which usually means a low voltage alarm of around 3.3v under load, for the best compromise between lifespan and performance.

...btw, how did you get notified of this discussion? I'm just curious.

GS88, 2018-04-05:
@Syonyk your blog is awesome. 

I would never design a BMS that takes lithium cell to 2.5V. But it is well within in spec for most 18650's. As you know increasing the cut off decrease the wear on the cells increases the number of "useful cycles". "Useful cycles" being a cycle capable of delivering the specified discharge rate at the specified capacity.

Yes, Li-ion and Lipo use the same base chemistry and share the same basic charge/discharge characteristics, But! they do have differences in their formulations that can give large trade-offs, between internal resistance, capacity and cycle life. 

A lot of care and design needs to go into a BMS to maximize performance and longevity of a given system. again trade-offs must be made. Engineering is all about trade-off's

Hobby Lipo packs and trade cycle life for raw discharge rates at reasonable capacities, and to preserve those discharge rates while still getting a reasonable number of cycles at those rates, require much more conservative cut-offs, using half the capacity at a time with a 3.3V cut off will more then double the useful number of cycles.

Brings me back to why I think the cutoff @Ralim has chosen is appropriate. 

Choosing an ideal cut off depends on so many parameters.

Its impractical to have a one size fits all solution. 

Choosing a very conservative cut off for hobby lipo's will not be optimal, but it covers the worst case and provides a safe option for all type of lithium cells.

If you want to get the most out of your cells you will need to run the iron in DC mode and provide your own battery management. 

As much as I would like the cut off adjustable in 0.1V increment, it's more confusing than its worth for most users.

Syonyk, 2018-04-05:
@JohnEdwa I monitor my referrer URLs for interesting looking discussion threads where stuff is linked, and this qualified.

I would suggest that "keeping the battery from fires and explosions" is a form of protecting the battery.  I consider a BMS to be a device for maintaining the battery (balancing as needed), as well as protecting the battery from defective external connections (a charger that's trying to overvolt the system, or a user that is trying to fully drain it).

Quadcopters are pulling an awful lot harder on the battery than a soldering iron.  This iron just doesn't pull that many amps.

@GS88 I would agree with your conclusion that a conservative cutoff is the best option.  But I'm certainly not the one who would be writing the patch for that.

Perhaps hide the adjustable cutoff in an "Advanced Settings" location?  Unless the flash is nearly full, that seems a useful option.

GS88, 2018-04-05:
Just FYI the stock firmware has an adjustable cutoff in the 9V-12V with 1V increments. 

The ADC on the STM32 chip controlling the iron is pretty good, but it does not use precision resistors to drop the voltage and the reference probably comes from a low spec linear regulator. The voltage readings could be off by up to 10%. You would need calibration for accuracy worth trusting as a BMS. 

I believe there is a calibration setting in @Ralim firmware but most people are not gonna use it.

I have skimmed over the settings section of the code to see if I could easily add something with my limited programming skills, but it looks like its probably more trouble then its worth.

I posted on this a few months ago asking for better voltage control but I realized I am better off with a $3  eBay BMS on my throwaway salvaged 18650 pack.

JohnEdwa, 2018-04-05:
@Syonyk 

> I would suggest that "keeping the battery from fires and explosions" is a form of protecting the battery.

Yeah, of course, but it's not their primary purpose by far.
As an example, let's take the common [DW01 protection IC](https://cdn.sparkfun.com/assets/learn_tutorials/2/5/1/DW01-P_DataSheet_V10.pdf), it's over voltage cutoff is at 4.250 ± 0.050V, and low voltage is at 2.40 ± 0.100V. At worst, it will let a cell be charged to 4.3V and then let it drain to 2.3V - doesn't sound like it's protecting the lifespan of the cell very well.   
I think we can both agree that in general if you trigger any of the BMS protections during normal use, something has gone wrong.

@GS88 
Heh, I never even read that far up, I just started from the "0.1V steps" comment.

In my opinion 3.3V per cell is a good limit for the iron, it's gentle enough for the cells while not leaving that much energy left. If you really desperately need to do one more joint, you can set it to DC and bypass the checks.

As for adding it yourself, the undervoltage check is done in [Main.cpp line 216](https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/main.cpp#L216), using the function from [Settings.cpp line 67](https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/Settings.cpp#L67), and the setting itself is done in [GUI.cpp line 272 to 285](https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/gui.cpp#L272). 
Right now all it does is sets the settings value to 0 (DC) or 1 to 4 (3S to 6S) in gui.cpp, calculates the proper minimum voltage for the cell count in the function in settings.cpp, and checks current voltage is larger than that in main.cpp.

To get voltage steps, replace the part in GUI.cpp with code copied from a temperature/time setting and modify the function to return that voltage instead of calculating based on cell count.

GS88, 2018-04-05:
@JohnEdwa I know, I just can't be bothered to set up the build environment :-P.

If your inclined, I would appreciate the ability to modify the per cell cut-off voltage in the settings menu.

Also battery protection module IC comes in all sorts of voltages cut-ins and cut-outs, for various design goals. The ones used in the cheap eBay one usually tend to maximize usable capacity over cycle life.

Most high capacity 18650 cells (2600mAh+) require 2.5 cut out and 4.3 charging to get their nominal capacity.  Almost all lithium batteries will tolerate that with decreased lifespan.

LarsSimonsen, 2018-05-07:
I agree with the need for more granular control of the cutoff voltage. Here is my use case: I use a laptop PSU that gives 19.3V under load. I have ordered a 6S battery pack. 

If I set the cutoff voltage for 6S, that gives me a 19.8V cutoff (at 3.3V per cell), so the iron will refuse the PSU. I believe 3.3V is a bit higher than it needs to be, so what if I set the cutoff voltage for 5S instead? That would give me a 16.5V cutoff, which is 2.75V per cell on a 6S pack. That is lower than I like. But at least I'd be able to use my PSU without having to switch back and forth between DC and 6S mode on the iron. 

Now, if I could set the cutoff voltage manually to 19.2V, that would give me a cutoff at 3.2V per cell, which I feel is perfect. I could even set it to 18.4V, in order to to get 3.1V per cell, if I decided that driving my cells a little bit lower would actually be worthwhile. Both these settings would allow me to seamlessly switch between battery and PSU.

I acknowledge that scrolling through hundreds of values will be a challenge though. Maybe the option to set a custom cutoff voltage per cell is the best one.

Berre1959, 2018-11-07:
I recently salvaged 14 Lipo packs (2S3P) from obsolete portable ticketing equipment, each pack containing internal protective balancing circuitry.  By putting three of those packs in series, I obtain four 6S3P power banks, capable of over 160 WH each.  Since I only use one bank at a time, three bancks remain at my disposal either in standby or charging.

closed 2023-06-05 by Ralim