Ralim/IronOS #51: Feature request: Autostart heat to sleep temperature

awigen, 2017-08-08:
It would be nice if the autostart setting was enabled there was also an option to startup in sleep mode.

1. Power is applied, soldering iron heats to sleep temp
2. Soldering iron is picked up and heats to work temp

Ralim, 2017-08-08:
Yep, can do :)

Ralim, 2017-08-09:
I have added this option in the latest tester version.
However, I don't think this work overly well as the accelerometer is rather noisy at startup as it settles into operation. Curious to see your thoughts on this.

Vidalcris, 2017-08-09:
What about using a delay at start when this option is enabled ?
The delay should be cancel if we press tip button to heat the iron.
10 seconds could be fine then ;)

awigen, 2017-08-09:
Masking the the accelerometer for the first 5-10 seconds should do the trick.

Ralim, 2017-08-10:
Yes I agree that masking the input for the first 10 seconds would work, 
Ill make up a testing build to test this soon.

Ralim, 2017-08-11:
@awigen @Vidalcris The latest push includes a mask on the accelerometer. Can you please test and check it works on your devices?

Vidalcris, 2017-08-11:
It work well on my side ;)

Ralim, 2017-08-13:
I'm going to close this for now as @Vidalcris has found it working, and it works on both of my units. 
Anyone can re-open this if there are any future issues with this.

closed 2017-08-13 by Ralim