Ralim/IronOS #52: Less sensitive timings for simultaneous button press?

kimble4, 2017-08-08:
I'm finding that pressing both buttons simultaneously is very unreliable.  Maybe it's my hands, maybe it's the quality of the switches, but about half the time I attempt to press both buttons simultaneously it is read as a press of whichever button happens to trigger first.  (This is particularly annoying if you want to get into the Extras menu and accidentally end up with the iron heating up.)

Presumably this is a simple constant that can be tweaked.  Not sure if it's worth having another settings option for...?

awigen, 2017-08-08:
I second that, I find the timing quite sensitive.

Ralim, 2017-08-09:
@awigen @kimble4 Can you test the latest release if it's easier for you?
It's hard for me to test this apparently :/

Vidalcris, 2017-08-09:
Way better imo ;)

kimble4, 2017-08-09:
Yep, that's solved it for me.  Brilliant!

Ralim, 2017-08-09:
Awesome! Ill close this, but open it if it gets harder again :)

closed 2017-08-09 by Ralim