Ralim/IronOS #53: Menu suggestions

Vidalcris, 2017-08-09:
I got two idea that maybe you would add ;)
-To be able to exit the menu by pressing boths buttons at the same time.
-in the menu : if pressing the button (USB side) it should stop cycle on main screen.

(Auto cycle stop is better imo if you want to choose between those two ideas ;) )

Ralim, 2017-08-09:
The first is trivial to add, the latter is not. So I'll add the first one tonight and then the other when I get a chance. :)

Vidalcris, 2017-08-09:
Thanks a lot man ;)

Trezona, 2017-08-09:
Am I correct in saying that by pressing both buttons here will also save any settings that you have just made before exiting the menu?

Ralim, 2017-08-09:
Yep :)
Haven't pushed a release yet that has this in it though, Will be doing that shortly.

Trezona, 2017-08-09:
Nice!
This will be great as I feel that with all these new features you will be implementing in the future the size of the menu structure will be rather large😉

Ralim, 2017-08-09:
Yeah, it is getting lengthy :/ 
I will be open to suggestions on a re-ordering if people want in the future too :)

Trezona, 2017-08-09:
Yes good idea. Have all the most important (needed) changes in front of the rest. This alows people to make these changes quickly and exit.

Vidalcris, 2017-08-09:
@Ralim just to be sure. Have u well received my donation? 

_Sent from my HONOR PLK-L01 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Ralim, 2017-08-09:
Yes indeed, thank you! 
I need to make a thank you file somewhere to let people know.

Vidalcris, 2017-08-09:
My pleasure, you deserve it ;)

Vidalcris, 2017-08-09:
I just got another idea to exit menu.
If we press usb button for more than 2s we come back to main screen and nothing happen until we release the button :)

Ralim, 2017-08-10:
I've already added the two button exit in the newer firmware, as this is by far the easiest option (and faster).

RE edit:
It was meant to be cooling 😛 But yes I can see this, I kind of want a middle ground, for it to mean "cooling" but the tip is still "hot" So dont really want to use HOT but might have to. Any further ideas ?

Vidalcris, 2017-08-10:
Yeah use HOT ;) its my latest idea lol

csdpm, 2017-08-11:
As a new user, the 'Cool' (I understand that it was intended to be 'Cooling') does seem a odd when, from the flashing temperature indication, it definitely isn't cool!

Given that the only way to restart at this point is by pressing the 'On' button, then what about 'OFF'?

Ralim, 2017-08-27:
I have added the two button exit in the 1.17 proper release.
It would be more effort to add the auto stop, but I can look into it if there is still demand?

closed 2017-09-17 by Ralim