Ralim/IronOS #56: Screen blinking super fast in sleep mode

Vidalcris, 2017-08-09:
When the temperature is under 65°C the screen start to blink very very fast.
That happen only on sleep mode (my delay is 10sec)

Ralim, 2017-08-10:
hmm, right. I'll fix this in a few hours.

Ralim, 2017-08-10:
... Or i can fix it now :P Can you test the newest release?

closed 2017-08-10 by Ralim

Vidalcris, 2017-08-10:
testing right now.

Vidalcris, 2017-08-10:
Where are u from ?

Vidalcris, 2017-08-10:
The screen is not blinking anymore on sleep mode or i did something wrong ?

Ralim, 2017-08-10:
The screen is only mean to blink in the cooling mode at the moment? The screen blinking on sleep was an accident.

Vidalcris, 2017-08-10:
ok so its working now ;)
(i contacted you on facebook if you are ok to add me ;) )

Ralim, 2017-08-10:
Yeah its all good now :)
I dont really use facebook that much these days :/

commit f0e1498e (2017-08-10): Fix screen blink

Fixes #56