Ralim/IronOS #58: Temperature setting

Vidalcris, 2017-08-10:
I think the first parameter in the menu should be the temperature setting.
Sometimes i use the iron to repair 3D models and the heating is so fast that i cannot change the setting temperature fast enough.. and so my tip is way too hot for my purpose at this moment and i have to wait until it cool down to my new setting.
With the temperature setting in the menu we could change the setting before heating ;)

Ralim, 2017-08-10:
I normally just use a sponge 😛 
I'm not really sure I want to provide two ways to change the temperature since it means that people will get confused as to how to set the temperature (trust me, they will).
Im happy to leave this here and see if others pipe up voting for this though, just dont want to add any more confusion to all of this :)

Vidalcris, 2017-08-10:
mhm ok the sponge is a good idea :p 
Let see what other people think about that !

Trezona, 2017-08-11:
I must say I prefer it the way it is. But then again I only use my iron for soldering.

Ralim, 2017-08-11:
@Trezona I haven't changed it yet. Going to let this one sit or a bit and see what people suggest.
I'm personally with you as well. (But open to what the community wants).

csdpm, 2017-08-11:
I've only had my TS100 a couple of days but I was a bit confused not to find a default temperature setting in the Settings menu when I first opened it.

Vidalcris, 2017-08-11:
Oh btw i dont say that the temperature parameter should not be available in heating mode.
I just ask for the temperature parameter to be added in the menu ;)

@csdpm thanks for your help :)

tigert, 2017-08-15:
Currently it remembers the last temperature you set it to, so you will need to change it if you do something different.

I guess if the last saved temperature would also appear as a "setting" as the first item of the Settings menu, it could help this case and the downside would be rather minor, as you don't normally* use the Settings menu very often.

(* we are not normal, we get excited about firmware updates and settings on a fricken soldering iron ;D)

Vidalcris, 2017-08-15:
Also @Ralim i would personnaly prefer to be able to stop the iron from heating will just pressing usb button, instead of the two button (there is way too much space between those button imo ...)
And then use both button to change temperature setting.

Like that i could just stop the iron each time i'm not using it instead of waiting for the sleep mode :)

Ralim, 2017-08-15:
@tigert Who said that we had to be *normal*.

I think I will add this as a menu option to the settings menu. Not sure how many people will use it, but cant really hurt to add it I guess.

@Vidalcris Can you split that out to a different issue :P

Ralim, 2017-09-17:
FYI
In the 2.0 fork (which will eventually become mainline) there is the ability to adjust the soldering temperature from the idle screen by pressing both buttons at once.

Ralim, 2017-09-28:
Correction, its on holding the front (tip) button. But working :)

closed 2017-09-28 by Ralim