Ralim/IronOS #59: Boot logo as screen saver ?

Vidalcris, 2017-08-11:
Is this possible to use the boot logo as screen saver.
This screen should appear only on the main screen (after 5sec delay or so ..)

At start the logo should stay forever until we press a button ;)

Ralim, 2017-08-13:
The OLED screen used on the iron can suffer from burn-in quite quickly. This was the main reason for the screen turning off automatically (it doesnt really use much power). 

However having the boot logo stay until you press a button makes sense. :)

Vidalcris, 2017-08-13:
Yeah i said screen saver but this is not really what i mean ;)
The screen should continue to turn off automatically but after showing the boot logo :)

Ralim, 2017-08-13:
Could be fairly easy to add it so that it shows when the unit is idle on the main screen (where the symbol of the iron normally appears) when its in-active until the screen turns off. Don't really want to cover up the screens you use when using it (sleep, shutdown, soldering) and in the other ones (settings, calibration) your actively using the unit. Thoughts ?

Vidalcris, 2017-08-13:
Yes i agree with everything you said ;)

csdpm, 2017-08-13:
> Thoughts ?

It would be nice to have a settings option to completely turn off the screensaver (the animated picture of soldering iron?) and have the iron power up straight to the Shut (Off) screen.

It would then be immediately possible to see the ambient tip temperature, calibrate it, calibrate the voltage etc. There is already direct access to the setting menu from there (you just can't get back). It would also become possible to (re)start the iron with a single button press rather than the current two (one to go to the screensaver, one to turn on).

Granted, it's not as pretty as having the animation but probably more logical and quicker to operate.

I guess this would probably sever the final visible link with original firmware.

Ralim, 2017-08-14:
Hmm this is interesting,
The Shutdown (Off) screen and the ambient temperature screens are technically different. However, I see where you are going with this. 

Currently, there is still only one press (as the bootup image is shown for fixed time). 
Would it work instead to have two options:
1. Replace the logo with the current tip temperature (Which would also disable the cooling/off/shut prompt)
2.  Option to disable all startup text & logos so you just straight away to that screen.

The reason I suggest this is it is a lot less extra code to support and should provide similar features?

I have no problem with links to the original firmware or not, I actually added that logo back in later on because I personally liked it. But I'm not overly attached to it 😛

csdpm, 2017-08-14:
Thanks. that sounds like a good compromise... I think.  I'm all for anything that saves you too much additional coding. :)

One of the reasons I mentioned going straight to the Off screen was that I hadn't (yet) found a way of getting to the Temperature Cal. that didn't involve some degree of (accidental) tip heating, ie. turn on, immediately do the two button turn-off to get to the Off screen (maybe I missed a route). The tip heats quite a lot in that time.

I was thinking that being able to do an immediate ambient temp cal. would be useful for tip changes. As I say, maybe I missed an existing route.

csdpm, 2017-08-14:
Edit: Found the direct path - 2 button press from the main screen. Sorry.

What you suggest sounds ideal then!

Ralim, 2017-08-14:
Thanks :)
The amount of code is not an issue, more trying to keep it simple so upkeep is easier. 😄 

Normally to enter temperature calibration I follow the following :
1. Press both buttons on the idle screen (shows the soldering iron picture / logo)
2. The unit will now show the current tip temperture
3. Press the button near the usb end, and you are in calibration mode :)

Also, you *really* don't need to do tip calibration very often. I normally do it only if I notice issues in the temperature response of the iron.

closed 2017-10-06 by Ralim