Ralim/IronOS #60: Offer: I could do a German translation

ErhardDu, 2017-08-11:
I bet I'm not right here but I'm new to github and so I don't know where to post this:
 I hereby offer to make a German translation if it's not too hard. If you tell me the source file I'll download it and modify.
Have a nice weekend!

Ralim, 2017-08-11:
Yep, you have the correct location :)
Issues as basically used to organize stuff that's happening (good or bad).

The file that holds the text is [strings.c](https://github.com/Ralim/ts100/blob/master/workspace/ts100/src/Strings.c)

If you can provide a translation of the text in the file (just copy and paste the English text and update where needed, then post back here). I'll make up a test build for you to check :)

ErhardDu, 2017-08-12:
```
#ifdef LANG_DE
const char* SettingsLongNames[14] =
		{
		/*These are all the help text for all the settings.*/
		/*All must start with 6 spaces so they come on screen nicely.*/
		"      Power source. Sets cutoff voltage. <DC 10V> <S 3.3V per cell>",
				"      Ruhetemperatur <C>", "      Ruhemodus nach <Minuten>",
				"      Abschaltzeit <Minuten>",
				"      Bewegungsempfindlichkeit <0.Aus 1.Minimal 9.Maximal>",
				"      Temperatur Einheit", "      Temperatur Runden",
				"      Temperaturanzeige Updaterate",
				"      Anzeigerichtung <A. Automatisch L. Linkshändig R. Rechtshändig>",
				"      Fronttaste für Temperaturboost einschalten",
				"      Temperatur im \"boost\"-Modus",
				"      Ändert \"Temperaturpfeile\" in Leistungsbalken",
				"      Automatischer Start beim Einschalten. J=Löttemp., R=Ruhemodus,N=Aus",
				"      Temperatur blinkt beim Abkühlen, solange noch heiß." };

const char* TempCalStatus[3] = { "Cal Temp", "Cal OK  ", "Cal Fehl" }; //All fixed 8 chars
const char* UVLOWarningString = "V gering"; //Fixed width 8 chars
const char* CoolingPromptString = "Kalt "; //Fixed width 5 chars
const char SettingTrueChar = 'J';
const char SettingFalseChar = 'N';
const char SettingSleepChar = 'R';
const char SettingFastChar = 'S';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'L';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'F';

#endif
```

Cpt-Huck, 2017-08-12:
Nice ! but actualy I like the eng lang more :P.. still will try the german version in the future since im from germany.

Ralim, 2017-08-13:
@ErhardDu @Cpt-Huck The latest release has the German option available for download. Could you please test and share any feedback? Hopefully, all the text worked in the font correctly. If anything didn't can you please let me know :)

Cpt-Huck, 2017-08-13:
#ifdef LANG_DE
const char* SettingsLongNames[14] =
		{
		/*These are all the help text for all the settings.*/
		/*All must start with 6 spaces so they come on screen nicely.*/
		"      Stromversorgung. Setzt Abschaltspannung <DC 10V> <S 3.3V pro Zelle>",
				"      Ruhetemperatur <C>", "      Ruhemodus nach <Minuten>",
				"      Abschaltzeit <Minuten>",
				"      Bewegungsempfindlichkeit <0.Aus 1.Minimal 9.Maximal>",
				"      Temperatur Einheit", "      Temperatur Runden",
				"      Temperaturanzeige Updaterate",
				"      Anzeigerichtung <A. Auto L. Linkshändig R. Rechtshändig>",
				"      Fronttaste für Temperaturboost einschalten",
				"      Temperatur im \"boost\"-Modus",
				"      Ändert \"Temperaturpfeile\" in Leistungsbalken",
				"      Automatischer Start beim Einschalten. J=Löttemp., R=Ruhemodus,N=Aus",
				"      Temperatur blinkt beim Abkühlen, solange noch heiß." };

const char* TempCalStatus[3] = { "Cal Temp", "Cal OK  ", "Cal Fehl" }; //All fixed 8 chars
const char* UVLOWarningString = "V gering"; //Fixed width 8 chars
const char* CoolingPromptString = "Kalt "; //Fixed width 5 chars
const char SettingTrueChar = 'J';
const char SettingFalseChar = 'N';
const char SettingSleepChar = 'R';
const char SettingFastChar = 'S';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'L';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'F';

#endif

@ErhardDu @Ralim 

minimal fixes in translation...

Changes:
		"      Stromversorgung. Setzt Abschaltspannung <DC 10V> <S 3.3V pro Zelle>",
				"      Anzeigerichtung <A. Auto L. Linkshändig R. Rechtshändig>",

Sounds strange:
const char* CoolingPromptString = "Kalt "; //Fixed width 5 chars
i don`t know if it fits in, but "Abkühlen" would be better.. mybe "Abkü"  ?!

maybe i`ll find some more things later on..

Ralim, 2017-08-13:
I've updated in the latest alpha to your translation. I have no idea which would work for the cooling prompt. Though in the next version I'll be adding a symbol, so we may actually just remove the text eventually.

Ralim, 2017-08-17:
@ErhardDu @Cpt-Huck 
Any further feedback on this or is it good to go?
Doesn't have to be perfect for now, just good enough for the first release, can always change/update this later on.

Ralim, 2017-08-26:
As far as I am aware this is done, so I'm closing the issue.

closed 2017-08-26 by Ralim

Cpt-Huck, 2017-08-27:
I'm  native German, but in the hobby of FPV drones, everybody uses english.. Banggood is in English, Betaflight an BLHeliSuite is English or most people use it with English language, because all the howto's are made in English language.. Also for me, English sounds better than German.. So I switch back to English.. Also the translation is just half finished.. All the short things when you enter the settings are still in English.. 

_Sent from my Samsung SM-G925F using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Ralim, 2017-08-29:
Hi @Cpt-Huck 
The translations were mostly just to be for the short settings names since it might be preferable for some users to have the reminders in their native language. There is no pressure to use the translations, however, they are easy to add so why not  😁