Ralim/IronOS #61: What is the font =] ??

kokein, 2017-08-12:
Hi Ralim, 

Can You tell me/us what font did you used ? :) 
It's very nice !

Ralim, 2017-08-12:
Uhhh, It's from a font pack I picked up somewhere online (Attached). 
[font.zip](https://github.com/Ralim/ts100/files/1219665/font.zip)

kokein, 2017-08-12:
Ok =) Thank You!
Nick

closed 2017-08-13 by Ralim

uncle-fed, 2017-09-12:
Make sure the font license allows its free usage and is compatible with GPL, if you intend to use GPL in the end. Licensing can get very nasty.

Ralim, 2017-09-12:
It was released for anyone to use freely in 2007. While no specific licence was attributed with the pack, GPL or MIT would be closest to what I interpret to be the original creator's intention as of when the original pack was posted. I do not know if the original poster of said font pack was the actual original creator but appears to be. The font itself has also appeared in many, many, other open source projects under GPL or BSD licences (including in ug8lib if I remember correctly).