Ralim/IronOS #62: Display blanks in Cooling mode while still too hot to touch

csdpm, 2017-08-12:
Hi,

I've noticed that the display goes blank in 'Cooling mode' before the tip has reached a safe temperature (before flashing stops). I guess this is a potential burn risk. (V1.17 Alpha 4).

I am using: Tip temperature 320C, Sleep timeout 1min, Sleep temp 200C, Shutdown time 5mins.

The solution might be to prevent display turn-off until the temperature is below the 50C flashing threshold. Maybe wait till 40-45C so the end of flashing can be seen. Hopefully that would still be high enough to allow the screen to blank in most climates.

As mentioned in another thread,  better alternatives to 'Cool' might be 'OFF' or 'Shut'.

Many thanks

Ralim, 2017-08-12:
Hi,
I've confirmed the issue here. The fix should be fairly easy. I'll push it out probably tomorrow. Thank you :)

I agree with the other thread and will be changing the cool in the next release.

csdpm, 2017-08-13:
Unfortunately extending the display timeout has introduced (re-introduced?) a bug.

I've had to turn off the temperature blink in the 'Shut' screen - that whole screen invert flashing is _horrible_ (sorry). With blinking turned off, the screen goes into rapid flashing between the time it was turning off previously and when it finally blanks. Moving the iron during this period restores steady display.

P.S. After seeing it, 'OFF' might still be better than 'Shut' - better size match with 'SLP'.

closed 2017-08-13 by Ralim

csdpm, 2017-08-13:
Excellent, thanks Ralim. Confirmed fixed... and I definitely like Off better :)

commit 4f0e3560 (2017-08-13): Fix #62
commit bb174c76 (2017-08-13): Fix #62