Ralim/IronOS #63: Calibrating tip temperature with boiling water?

ErhardDu, 2017-08-12:
Just tried to calibrate temperature with boiling water as this may give the most precise way to do it unless you live on Mount Everest. I noticed that this is not possible and gives an calibration error.
Is it possible to change this behavior?

Ralim, 2017-08-12:
The tip calibration is for calibrating offset, not actual temperature, as the thermocouple used has a fairly well controlled response curve. 

So to use the temperature calibration you calibrate when the tip and the unit are at room temperature. (there is a temperature sensor inside the handle that is used for this as well as cold Junction compensation)

ErhardDu, 2017-08-12:
OK thanks :)

closed 2017-08-12 by ErhardDu