Ralim/IronOS #64: Adding logos to the sleep / shutdown screens

Ralim, 2017-08-13:
This was originally started by @Vidalcris.
Adding logo's as an option instead of the text on the shutdown and sleep screens instead of the text.

This will probably be a 2 part issue. Firstly getting the image showing and working, and secondly having smaller text font to show a warning next to the image.

Vidalcris, 2017-08-13:
Oh that will be an option on the menu ?
Next step is a software to modify the theme :p

Ralim, 2017-08-14:
That was my plan, as some users may prefer the text. (And it's very little extra code).

Maybe one day, that would be alot more work :)

EricJanVanDenBogaard, 2017-08-19:
About adding a logo... is maybe not that important, but maybe the option to enter a ID text/name could be useful if one is soldering at a workshop with people who also have a TS100 and soldering there. some tools are just a common thing :)

Ralim, 2017-08-20:
This is more for being able to show logos/images on the screens such as the cooling down screen (like a heat warning). The current method of showing a startup logo will stay as probably the best method for branding your unit apart from changing the case :)

EricJanVanDenBogaard, 2017-08-20:
Okay.

closed 2017-09-28 by Ralim