Ralim/IronOS #65: Sleep mode not working on v1.17-alpha5

Vidalcris, 2017-08-13:
I even tried to reduce the sensibility to 2 but this is not activating sleeping mode anymore.

Vidalcris, 2017-08-13:
Mhm weird i solved the problem by installing the original firmware again then come back to your..
My accelerometer was not working correctly anymore. Even with Alpha4.

If the accelerometer is working like in my quadcopter we should be able to calibrate him, right ?
Also even at Msense = 9 i need a pretty big movement to get out sleep mode.

Ralim, 2017-08-13:
Hi, 
Did you power cycle the iron after changing the sensitivity? At the moment It only changes on power up. (that *is* something I need to fix as people forget). 

It is only an accelerometer (no gyro and no magnetometer) so there is very little we can calibrate as there is no combination filter to tune (ie madwick or kalman). Furthermore the software actually uses a built in sensing function to the accelerometer for movement detection. So what the tuning numbers do is set the threshold for this internal motion sensitivity.

Vidalcris, 2017-08-13:
Yes i tried to restart the iron.
I had to hit the ts100 to be able to get out of sleep mode even at 9 for the sensitivity.

About the accelerometer this is what i think is weird, because imo it need to calibrate to know when he is not moving.
They should have used a gyro instead :-/ (oh btw the accelerometer allow the screen orientation ;) )

Ralim, 2017-08-13:
That's... Very odd. Out of curiosity have you opened your iron at any point in time? The one time I had issues with the sensitivity was the small inner board actually coming loose (it caused other issues too).

A sensitivity level of 9 the IMU is nearly as sensitive as possible. Which is one limitation for this application that if you move too small we just can't measure it above the noise level of the accelerometer?
A gyro is way, way worse to use imho (I'm dealing with one now in another project), they to be significantly noisier. An ideal solution would be a gyro+accelerometer such as the mpu series. but they cost more :P

Yeah the accelerometer has built in orientation detection we use :)

Vidalcris, 2017-08-13:
I have not opened my ts100 and i received it 10days ago, so its a new unit ;)
I'm pretty sure that the accelerometer was in a sort of bug..
I restarted many time and as i said, i had to hit pretty hard the iron to get out of sleep mode.

That happened just after i flashed the latest Alpha 5. at this point i was unable to get into sleep mode.
Then i flashed Alpha 4 and i got the problem to get out of sleep mode.
Then i flashed original firmware then again flashed Alpha 5 and that was working correctly.

Maybe this is just a rare problem happening .. maybe because of too much flashing..
Anyway i'm not able to reproduce it ;)

Vidalcris, 2017-08-13:
Ok the problem happen again .. i was using msense =9.
Now even if i go back to 7 this is not working.

Ralim, 2017-08-14:
Hi,
It's interesting, I have noticed the accelerometer sometimes does not behave well at high sensitivities, however, it usually gets locked into "movement occurred" mode rather than not triggering movement.

Can you try a force-reset of your settings (hold the button near the USB while booting until it tells you it reset) and then power cycle? 

I currently don't have an active hex for debugging the accelerometer, however, I can make one if this issue persists. I cannot replicate this issue at all locally :( Which makes me think its not something related to flashing all the different versions. 

The accelerometer has 127 steps for the threshold for motion. 
Currently the translation from the setting to the threshold is :
```
uint8_t sens = 9 * 6 + 5;
	sens -= 6 * sensitivity;
```

So it should have no issue with settings higher than 7 :/

If force re-setting the settings does not help, I can make up a debug firmware for you.

Vidalcris, 2017-08-14:
Hi :)
So i just tried a reset then msense = 8 the problem is the same.
Then i changed back to 7 and restart.
It work but when in sleep mode i have to do a fast movement to reactivate.
If i move slowly i have to do a 180° to reactivate ^^

Ralim, 2017-08-14:
Damn :/ This might be an issue in your accelerometer. Ill make up some testing firmware when I get a chance. :/

Vidalcris, 2017-08-14:
Oh dont worry, anyway if there is a problem with the accelerometer that can not be solved a debug will maybe not change anything...
But ^^ i'm pretty sure it is working correctly with original firmware.. i will try again right now ;)

Ralim, 2017-08-14:
The original firmware does not use the built in motion detection, but instead reads the value of the acceleration values. This means there is a completely different algorithm. I'll make up a testing firmware that shows the threshold on screen and shows when motion was tripped.

Vidalcris, 2017-08-14:
oh ok this is weird then .. am i the only one to have this problem ? ^^
Is the method they use more complicated to code ? :p

Ralim, 2017-08-14:
The method they used would not work correctly in some situations (never let the iron sleep), and it also slowed down the entire system as it polled the unit for measurements really often.

Basically what they did was looked to see if the acceleration in the axis changed significantly, which meant it took more deviation to trigger a wakeup, and because of the polling nature it would completely miss events that occured between the measurement points.

In my system, I use the built in motion detection in the sensor (MMA8652FC), and this is setup to trigger the INT line to the mcu. This means that the interrupt in the mcu stores the event instantly and we cannot miss events. However, this also means that the entire system of detection is done in the IC and we don't have much transparency into this. For more details, you can try reading their datasheet :) (Google the part number).

Ralim, 2017-08-14:
If you are reading the datasheet, chapter 5.6 "Freefall and Motion detection" is the part of the chip we use. The sleep/wake-up is just used to save power in the accelerometer which we don't use.

Vidalcris, 2017-08-14:
I see that there is offset that we can modify OFF_X, OFF_Y,OFF_Z.
That could maybe solve my problem ? :)

Ralim, 2017-08-14:
The IC automatically uses all 3 axis to detect movements for us, as it looks for a change in the measurements. I.e. if any of the axis change by more than the threshold amount for a period of time it will trigger the movement detection.

The absolute offset of any of the axis does not actually apply in this case as we are only looking at the change per period time. 

One option would be to reduce the filtering time that is used, however making this too much smaller makes the unit prone to waking up wayy too often when its still.

Kenneth-N, 2017-08-19:
+1
I have the same issue.
With motion sense on 7, it works fine, set to 8 it never sleeps.
Nice fw by the way.

Ralim, 2017-08-20:
Good to have a second person with the issue.

I will tweak the filtering time in the next release to see if this improves the issue :)

Also @Kenneth-N & @Vidalcris, what power source are you using to power your irons?

Vidalcris, 2017-08-20:
24V 4A or lipos 4S.

Kenneth-N, 2017-08-20:
18V Li-Ion 5S (Makita 18V battery pack).

thanks4opensource, 2017-08-22:
Just another confirmation of strange sensitivity/accelerometer behavior: I've had times (using Ralim 1.16.3) where the iron wouldn't go into sleep mode despite no motion occurring. Also, subjectively seems like there's a difference between wake-from-sleep sensitivity and flip-left-right-upsidedown-LCD sensitivity. I've been using 7, 8, or 9 sensitivity setting (higher the better; I have no problems with spurious wakeup, and want intentional wakeup when gently picking up the iron) but have had times when the LCD flipped and required a violent shaking to flip back (again at 7,8, or 9 setting). Maybe just noise/flakiness in the hardware. And @Ralim, I know what you mean about gyro data being noisier than accelerometer or magnetometer -- I've seen it in my own code, on other hardware.
But the LCD thing not a problem. I think I'll go to fixed right-hand instead of automatic flipping. Just wanted to mention it in case of any interest.

Ralim, 2017-08-22:
Hi,
The screen rotation and the movement detection are completely separate. The screen rotation has a much, much longer filter period on it and requires a significant rotation to flip. This is deliberate so that the screen will keep its current rotation when the screen is leveled facing up (ie if you solder with screen straight up, rather than at the typical ~30 deg angle. When getting the screen to flip, it works like a phone. So you just hold it the way you want for a little bit and it will pick it up after a bit. Moving rapidly will cause the filter to filter it out and it usually won't trip easily. Once I have my new workspace setup I will be doing another update video on how this all works :)

I need to look into the filtering times for the accelerometer, I have managed to reproduce this on one of my irons now so I can test with it.

thanks4opensource, 2017-08-22:
@Ralim wrote:
> When getting the screen to flip, it works like a phone. So you just hold it the way you want for a little bit and it will pick it up after a bit. Moving rapidly will cause the filter to filter it out and it usually won't trip easily.

Thanks! Just tried it, and of course you're absolutely right. Works perfectly once one understands it. I would get frustrated and think, "The screen's not flipping, maybe the accelerometer's  not giving a strong enough signal, I'd better shake it hard to make it flip." Totally the wrong thing to do.

Never underestimate a user's ability f*** things up.  ;) ;) ;)

Ralim, 2017-08-22:
No problem at all, The feature I'm using is built into the accelerometer and is what phones also use, so it works in exactly the same way :)

The actual issue with the accelerometers I will look into though :)

Ralim, 2017-09-28:
In the new firmware, the motion sensitivity is retuned and appears to be working a lot better.
Re-open this (or comment) if you are still having issues, anyone 😁

closed 2017-09-28 by Ralim