Ralim/IronOS #68: left hand orientation and settings image is confusing

tigert, 2017-08-15:
The left / right hand screen orientation and the settings image ("press this button for settings") image is confusing because it is flipped compared to the iron you are holding on your hand.

So while it is correct in the image itself, the tip of the iron could point to the actual tip of the iron, so it would not be backwards and you wouldnt accidentally press the right button where the arrow points to...

This might be related to #64 since one could perhaps flip the graphic if the text is drawn on it separately.

Ralim, 2017-08-15:
This is indeed linked into #64, as it will need to come after I make up the smaller font 😄 

I'm surprised this wasn't an issue already, it must have been mentioned somewhere as I thought I've touched on this already. But anyway, yes this is on the road map for after #64 is completed. I intend to polish up a fair few of the screen writing code functions so updating this image at the same time makes sense.

tigert, 2017-08-16:
I can help with the font, I like pixels :-) But the format is mysterious for me.
How does one define fonts?

Ralim, 2017-08-18:
I have a reference font I'll currently go with most likely. So mostly I just need to sit down and encode it. 😄 

FYI The format for the pixels is : 
The screen is actually treated as a 8 bit high x 192 pixel wide strip. Where each byte in order is one 'strip' of pixels down the screen. The Most Significant Bit (MSB) of each byte is the top most pixel, then the next bit (7th) is the next pixel down etc.

So the font encoding is 12 bytes that are the 12 'strips' for the top half of the character, followed by another 12 bytes that are the bottom half.

No tool really exists, I have a very hurriedly coded one I use for drawing the characters out that simplifies it a bit :) But isn't perfect by any means.

Another reference for this weird format would be the source for the boot up logo editor as well.

I need to write more documentation on this at some point. :/

Ralim, 2017-10-01:
Closing this as this is being addressed in the graphics for 2.x issue

closed 2017-10-01 by Ralim