Ralim/IronOS #69: Buttons usage

Vidalcris, 2017-08-16:
i would personnaly prefer to be able to stop the iron from heating will just pressing usb button, instead of the two button (there is way too much space between those button imo ...)
And then use both button to change temperature setting.

Like that i could just stop the iron each time i'm not using it instead of waiting for the sleep mode :)
What are you all thinking about that ?

csdpm, 2017-08-16:
I have to admit, I'm rather hamfisted with the two button press too.

Ralim, 2017-08-17:
Thanks for splitting this out @Vidalcris.

A few people have had issues with the two finger button press, so this is definitely an issue.
However, I would rather not force a button change on users of the iron. 
Would having a setting to swap the button functions work (ie swap double button press and the single B button press)? The alternative is supporting long button pressing the B button to exit but that's more work.

Thoughts? Or is there an even better solution?
I have thought about using the tap detection in the accelerometer to provide a means to enter temperature adjustment, but im not sure how annoying that would be?

Vidalcris, 2017-08-17:
Yeah if you think users need to change temp often, this is not my case and i think most of the users are not changing temp all day long ;)
Anyway an option in the menu will work.

Tap detection is not a serious option imo..

I think the menu should have submenu soon :p

Trezona, 2017-08-17:
I also don’t change the temperature at all. I just use the "Boost" function every now and again.
I have also got so used to using the 2 buttons for powering off, this is really not an issue for me at all.

Vidalcris, 2017-08-17:
I dont know how u can use both button with no problems. And my hands are big..
I'm thinking to 3D print button with a better design...

I dont understand why those buttons are so far away ... even the original firmware use both buttons :-/
This is a flaw in the design imo..

Ralim, 2017-08-17:
I'm happy to add a menu option to effectively swap the functions :)
Tap detection was a joke (though possible technically).

I normally just use my thumb over both buttons to press them at the same time, or both hands. But it's not too hard to swap (I'll default the option to off).

Ralim, 2017-09-28:
In the 2.x fork you can now long hold the rear button (usb end) to exit the soldering mode as well as the double press.

csdpm, 2017-09-28:
Much better. Thanks :)

closed 2017-09-29 by Ralim