Ralim/IronOS #70: Temperature on Startup screen

karatektus, 2017-08-17:
I've been using this for  a while now and im loving it so far. One issue i ran into multiple times was this:

Im done soldering and need my Battery for something else. so i unplug the iron. A short time later im done and want to pack up, but I don't know how hot the iron is. If i now power it up again it doesnt show the temperature but the graphic which tells me how to start soldering. 

It would be really useful if it could show the temperature on this screen.

Thanks :)

Vidalcris, 2017-08-17:
U can press both button on main screen to see temp :)
But i agree that could be cool to see it directly on main screen.

karatektus, 2017-08-17:
oh thanks!
yeah it would still be usefull. if you miss one button it would heat up :/

EricJanVanDenBogaard, 2017-08-17:
I agree with karatektus, this would be a good/necessary feature, it would  also be a warning when tip is hot, or use it as a feature, when power is interrupted, or when making ready for transport, and store equipment in backpack.

Vidalcris, 2017-08-17:
In fact the actual main screen is useless so we could just show some information like : 
-Actual temperature (with danger logo if higher than 50°C)
-Heating temperature setting
-Boost setting...
-Even more .. but look like a lot of people dont want small fonts :-/

EricJanVanDenBogaard, 2017-08-17:
a auto cycle through (as temporary screensaver) of this info would be possible ? with option to switch on by button press.

Vidalcris, 2017-08-17:
Everything is possible .. the hardest thing is to found something ok for everyone :-/
Here is approximately what i mean : https://ibb.co/eVLtuv

EricJanVanDenBogaard, 2017-08-17:
I must say i like the current font, and would prefer a cycle through per item on the screen instead of adding a smaller in-screen font, but i guess that's personal taste of me.

karatektus, 2017-08-17:
I wpuld like to always be able to see the current temperature. But you could show that on the legt side and something else on the right.

Ralim, 2017-08-18:
Hi,
I would have to agree with @EricJanVanDenBogaard that smaller fonts cannot be the only method used. I'm okay with extra information being shown in a small font, but cannot use small fonts for critical information (such as the current tip temperature).

I like the design by @Vidalcris with the logo setup and the text to the side. It would make a fairly decent replacement for the current idle logo screen for some users.

This will also be blocking on #64 which is, in turn, blocking on the v1.17 release. (Which should be coming soon, just testing at the moment). Just a bit tied up at the moment :/

EricJanVanDenBogaard, 2017-08-19:
maybe have the display showing tip temperature when TS100 is vertical with tip up ?

Ralim, 2017-08-20:
@EricJanVanDenBogaard That would definitely be possible to achieve. Should be a fairly easy change to add if the orientation detection in the accelerometer works for this.

Could you maybe break that out into a separate issue to implement? Since I think adding it is separate to this one?

EricJanVanDenBogaard, 2017-08-20:
okay.

Ralim, 2017-09-28:
In the 2.x fork if you enable advanced screens the temperature is shown live on the main screen now :)

closed 2017-10-06 by Ralim