Ralim/IronOS #71: Add lock mode to disable buttons

erpalma, 2017-08-17:
Hi, thanks for the nice work!

I'd like to suggest you a new feature: lock mode. Sometimes it happens that I press buttons during soldering, usually when soldering in bad positions. It would be nice to be able to lock the buttons by, e.g., keeping one pressed for 5/10 sec, same for unlocking. Of course a on screen "LOCKED" message is necessary when a button is pressed.

Ralim, 2017-08-18:
Hi,
This is a very logical feature to add if users are running into this issue 😄 
Currently this is not possible to implement until I re-work the button handling code a bit further to handle long button press detection (#72). Which is currently aimed for the 1.18 release.

This would probably end up being a feature on long press of the B button (the one near the usb end). Would that work ?

EricJanVanDenBogaard, 2017-08-19:
Long button press instead of the simultaneous button press ? that would be great ! you can even multiply functionality by giving separate functions to the tip and usb buttons that way.

erpalma, 2017-08-23:
Yes I think that would work perfectly!

Ralim, 2017-10-06:
Hi @erpalma 
So at the moment, long press on the back button (usb) is used for exit.
Would you still want this feature? If so how would you activate it?

closed 2017-12-12 by Ralim

chick0n, 2018-11-20:
Is this feature implemented now? Would be nice....