Ralim/IronOS #73: Display tip temperature when holding tip up

EricJanVanDenBogaard, 2017-08-20:
To have the display show tip temperature while holding the TS100 vertical with tip up,
would be a handy feature, this feature should be available in all modes also startup/idle mode,
(also when just plugging in from a short disconnection)
the vertical & tip up position would not give conflict with other functionality.

av8er, 2017-08-21:
Why make this so complicated? If the iron is plugged in with the temperature above 60c or whatever value we pick, just have it display the temp until it's below.

Ralim, 2017-08-21:
I believe the intent behind this was to provide an easy way to view the temperature no matter what temperature it is at?

EricJanVanDenBogaard, 2017-08-27:
didn't thought this one over i guess av8er is right, that should be enough. close this one,
does the voltage still need to be calibrated for the cell protection voltage, when setting the cell count ?

closed 2017-08-29 by EricJanVanDenBogaard