Ralim/IronOS #75: Sub-1-minute sleep, longer version display  enhancement

thanks4opensource, 2017-08-22:
My wish list, in order of importance:

Sub-1-minute sleep: The TS100 heats up so fast it's up to temperature by the time I move it from my homemade stand to the PC board. No need to be oxidizing the tip for full minute before going to SLTMP. (Although if I had a hardware wish list it would be for a piezo buzzer to beep when up to temperature.) Are there enough characters on the LCD to display "SLTME 01M" (etc) and then continue down through"55S", "50S", "45S" ... "05S"? Unless ... there's a hardware reason for the 1 minute minimum (burn out the tip's heater by cycling it from SLTMP to working temp too many times?)

Much less important:

The version string displayed at power-up disappears so quickly it's hard to read. Suggest two seconds before going into the ON/SETUP graphic/animation. But ... only if the firmware starts listening to button pushes while the version string is still displayed (nobody wants to have to wait 2 seconds before being able to initiate heat-up).

As always, thanks for the firmware, and for being open to suggestions regardless of whether you decide to implement them or not.

Ralim, 2017-08-22:
Hi, 
Hopefully I can help with these :) 
For your first requirement, it already exists! Keep cycling through the options and it will give you the options of 10,20,30,40,50 seconds as well :smile:

I'm not sure what the cycle rating on the tip is, but quite a few people use these as their daily use irons and haven't had any issues (and tips are cheap really). 

I reduced the display time to be really quick as most people do not pay it much attention usually. At the moment you can hold the B button on start-up and to reset settings. I could make the firmware display the version number for longer if you are holding this button as well. Since the software shows the reset prompt and only resets if you have been holding the button after a 1 second delay, so you would just release when you see the reset prompt to avoid actually resetting the settings.

thanks4opensource, 2017-08-22:
Thanks, @Ralim. Did you add the sub-1-minute values starting with the 1.17 alphas? I just now tried it again and 1.16.3 for me increments from 01 to 30 and then cycles back to 01 and starts again. All using the front/tip button while in SETTINGS with the SLTIM option displayed. Maybe I'm doing something wrong.
 
I had another thought about the version string: Maybe don't display it at startup but have it as an un-settable option in the settings menu? Like the standard Help -> About menu in desktop app menubars. That way those disinterested (99% of users, 99% of the time) wouldn't even have to see it flicker by -- much less wait for it -- but it would be available for checking as desired.

Vidalcris, 2017-08-22:
You should try latest release before asking improvement ;)
btw those modifications are available in the latest alpha.

Ralim, 2017-08-22:
@thanks4opensource 
Sorry I completely forgot to mention it's in 1.17.x versions. The 1.17 proper release should be out soon when I have a chance to push it up :smile:. 

Putting version string in the settings menu could work, and shouldn't be too hard either. Hmm, that might come soon :)

thanks4opensource, 2017-08-22:
Thanks for the info, @Vidalcris and @Ralim. I'll switch to 1.17 when the proper release is up.

Ralim, 2017-08-26:
@thanks4opensource The latest 1.17 release is up.

thanks4opensource, 2017-08-26:
Thanks, @Ralim. I'll be flashing it and enjoying the sub-1-minute sleep times!

Ralim, 2017-09-28:
I'm going to close this as this is in the 1.17 release and the 2.x release :)

closed 2017-09-28 by Ralim