Ralim/IronOS #76: Bug report: Error temperature display when off and cooling down V1.17

csdpm, 2017-08-26:
Sorry to come up with a bug report so soon after the V1.17 release.

When the iron is 'Off and cooling down (flashing enabled) I've found a strange temperature display jump. Flashing stopped as expected at 50'C and the temperature display continued to fall. Once the temperature display hit 37'C I decided to touch the tip and thought 'ouch that still feels pretty hot!'.

Watching the display further I found that once the temperature display hits 27'C, it suddenly jumps back up to 49'C and then continues down again.

That looks a bit of an odd one! It seems repeatable.

Ralim, 2017-08-26:
This is known as an issue that currently occurs.
The last part of code from the original software from miniware was the adc -> temperature conversion formula, which does this because of the way it seems to do compensation. You can also notice this if you watch really carefully as the iron heats up.

As their formula appears to be accurate at both room temperature and the soldering temperatures I have not yet tried to re-write it. Thus this occurs. :)

csdpm, 2017-08-27:
Thanks Ralim. At least it's a known problem then, I guess there must be an accumulating error that they have to 'fix' before getting down to ambient.

The workaround is simple anyway, don't deliberately touch the tip. :)

Ralim, 2017-09-02:
Just letting you know this has been patched in the coming 2.0 re-write (complete rewrite of the temperature measurement code fixed this issue completely)

Ralim, 2017-09-28:
@csdpm Letting you know 2.x alpha is up if you feel like testing it :)

csdpm, 2017-09-28:
Thanks Ben. Yes, the temperature looks fine now. :)

closed 2017-09-29 by Ralim