Ralim/IronOS #79: Moskito bite shocker

Cpt-Huck, 2017-08-28:
Is there any way to get the temperature constantly to 51 degrees?

Because always when I fly my quads an moskito storm is incoming and I get like 5 bites in a minute..

Moskito spray didn't work BTW 

To prevent the moskito bites getting bigger, you need to apply heat to the bite..
And I remember one device which heats up to 51 degrees..

I think my english is brilliant ¯\_(ツ)_/¯

Okay maybe if it's possible, make the temperature range from 50 to 450 degrees.. That would be great! 😊

Ralim, 2017-08-29:
Hi,
Technically there is no issue to doing this, except for the current unresolved issue of the thermocouple temperature curve code that is a leftover from the original code. When I re-profile the thermocouple code to work correctly at low temperatures, this would be completely doable. 

At the moment I could drop the lower limit down, however, I do not think that the temperature will be very stable.

I can test if its stable, if it is I will just change the limit in the next release, and if not it will need to wait until I re-write the thermocouple code 😄

dbrgn, 2017-09-05:
Basically emulating [this](https://www.amazon.de/bite-away-Elektronischer-Schwellungen-Insektenstichen/dp/B003CMKQTS/). Great idea :)

Ralim, 2017-09-05:
I have re-written the thermocouple code in the coming release that should allow for this to be added as a mode to the iron :)

tigert, 2017-09-07:
I am joking, but I'm just saying that maintaining a consistent 56.5C on a water bath will result in perfect medium rare Sous Vide beef steak...

Ralim, 2017-09-08:
You joke.. But would be tempting to try just to see 😀

dbrgn, 2017-09-08:
You joke... But a sous-vide cooker for 50$ ain't bad :laughing:

tigert, 2017-09-13:
Well. Yeah. And I was also planning to do a sous-vide cooker with an arduino.

Ralim, 2017-09-28:
In the 2.x fork now you *should* be able to do this, however at the low temperatures the response is a little jumpy because of the amount of power being delivered to the tip, so be careful with the temperature before you just apply it 😄 (*I'm not responsible if you burn yourself*). I would suggest using a lower cell count battery to reduce the jumping of the tip at the low end.

In regards to the Sous Vide, you could try the tip at 50 or 60C and see what happens? 😀 Might work...

The newer 2.x release is more stable because of the system being used for the drive [see here](https://ralimtek.com/STM32_Double_PWM/).

closed 2017-10-28 by Ralim