Ralim/IronOS #80: Problem with MSENSE 8 and 9

thanks4opensource, 2017-08-30:
On my TS100, MSENSE 8 and 9 seem to be "infinite" sensitivity. Can someone test the following on theirs and confirm or disprove this?

I would like the iron to stay awake during normal soldering, but go to sleep very quickly when stationary in its stand. I'm using these settings:
 ```
PWRSC DC
STMP 050
STME 10S
SHTME 03	
MSENSE 7	
TMPUNT C	
TMPRND 1
TMPSPD M
DSPROT R
BOOST F
BTMP 400
PWRDSP T
ASTART F
CLBLNK T
```

With "MSENSE 7" it's too "insensitive" -- it goes to sleep during normal soldering motions and requires a significant shaking to wake up.

With "MSENSE 8" (or 9) it never goes to sleep or shutdown. I've left it motionless in the stand for 15 minutes and it stays on. The stand and workbench are rock-solid, no movement at all.

There was a small change in the MMA8652FC.c accelerometer initialization between 1.16.3 and 1.17, but I can't remember if I had the problem with the earlier release. Also a change in Modes.c, but that just affected timeout values.

Thanks for any help.

Ralim, 2017-08-31:
Hi,
This is a duplicate of #65 
This is on the roadmap to be fixed in the next version as soon as I can track down whats causing it :)
The problem was worse in the earlier release, however the issue here is basically a trade off between bottom end sensitivity and noise wakeups

thanks4opensource, 2017-08-31:
Thanks, @Ralim. Apologies for the duplicate. I searched unsuccessfully for "MSENSE" but should have searched for "sleep". Or not forgotten that I had already commented in #65. ;)   My bad. Closing this. Will test alphas as soon as you come up with something to try.

closed 2017-08-31 by thanks4opensource