Ralim/IronOS #82: Bulgarian translation

gkrystev, 2017-09-07:
As there are some translations already available, I propose a translation for Bulgarian language
Here is the code with the translation.

```
#ifdef LANG_BG
const char* SettingsLongNames[14] =
		{
		/*These are all the help text for all the settings.*/
		/*All must start with 6 spaces so they come on screen nicely.*/
		"      Захранване. Настройка за минимално напрежение. <DC 10V> <S 3.3V за клетка>",
		"      Температура при \"режим на сън\" <S>", "      Време за включване в \"режим на сън\" <минути>",
		"      Време за изключване <минути>",
		"      Чуствителност за движение <0.Изключено 1.Слабо 9.Силно>",
		"      Температурна скала", "      Закръгление на температурата",
		"      Скорост за опресняване на температурата на дисплея",
		"      Ориентация на дисплея <A. Автоматично L. За лява ръка R. За дясна ръка>",
		"      Включване на предния бутон в режим за временно повишаване на температурата (до макс. 450C) при работа",
		"      Температура в режим на временно повишаване",
		"      Показвай \"индикатор на напрежение\" вместо \"стреки за температура\" на дисплея при работа",
		"      Режим на поялника при включване. T=Работен, S=Режим на сън, F=Изключен",
		"      След изключване от работен режим, индикатора за температура да мига докато човката все още е топла" };

const char* TempCalStatus[3] = { "Кал Темп", "Кал ОК  ", "Кал Греш" }; //All fixed 8 chars
const char* UVLOWarningString = "Нис Напр"; //Fixed width 8 chars
const char* CoolingPromptString = "Студ "; //Fixed width 5 chars
const char SettingTrueChar = 'T';
const char SettingFalseChar = 'F';
const char SettingSleepChar = 'S';
const char SettingFastChar = 'F';
const char SettingMediumChar = 'M';
const char SettingSlowChar = 'S';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';
const char SettingTempCChar = 'C';
const char SettingTempFChar = 'F';

#endif
```

Ralim, 2017-09-07:
Hi,
Thank you for the translation :smile: 
The current font does not have all the symbols used, but more can be added :)

This might take a little bit to get added as I will need to expand the font if that's okay?

gkrystev, 2017-09-07:
Yes, that's fine :)

uncle-fed, 2017-09-25:
This also should be merged with issue #87 where I am working on Cyrillic font that will cover Russian, Ukrainian, Belorussian, Bulgarian and other Cyrillic based alphabets.

Ralim, 2017-12-12:
@gkrystev Any chance you can update this to the newer format? (has more menu items)

gkrystev, 2017-12-13:
@Ralim yes, of course, will do it

gkrystev, 2017-12-13:
Hi, 
Here is the updated code.
For the True/False/Right/Left/Auto chars I have added some options like using Unicode characters or some existing ASCII characters. If they are not supported, you can left them as default ones.

```
#ifdef LANG_BG
const char* SettingsLongNames[16] =
		{
				// These are all the help text for all the settings.
				// No requirements on spacing or length.
				/* Power source (DC or batt)          */"Източник на захранване. Минимално напрежение. <DC 10V> <S 3.3V за клетка>",
				/* Sleep temperature                  */"Температура при режим \"сън\" <C>",
				/* Sleep timeout                      */"Включване в режим \"сън\" след: <Минути/Секунди>",
				/* Shutdown timeout                   */"Изключване след: <Минути>",
				/* Motion sensitivity level           */"Усещане за движение <0.Изключено 1.Слабо 9.Силно>",
				/* Temperature in F and C             */"Температурна скала <C=Целзии F=Фаренхайт>",
				/* Advanced idle display mode enabled */"Детайлна информация със ситен шрифт на екрана в режим на покой",
				/* Display rotation mode              */"Ориентация на дисплея <A. Автоматично L. Лява Ръка R. Дясна Ръка>",
				/* Boost enabled                      */"Ползвай предния бутон за \"турбо\" режим който температурата до 450C при запояване",
				/* Boost temperature                  */"Температура при \"турбо\" режим",
				/* Automatic start mode               */"Режим на поялника при включване на захранването. T=Работен, S=Режим на сън, F=Изключен",
				/* Cooldown blink                     */"След изключване от работен режим, да мига индикатора за температура докато човката на поялника все още е топла",
				/* Temperature calibration enter menu */"Калибриране на температурата",
				/* Settings reset command             */"Връщане на фабрични настройки",
				/* Calibrate input voltage            */"Калибриране на входното напрежение. Задръжте бутон за изход",
				/* Advanced soldering screen enabled  */"Детайлна информация в работен режим при запояване", };

const char* SettingsCalibrationWarning =
		"Уверете се, че човката на поялника е със стайна температура преди да продължите!";
const char* UVLOWarningString = "Ниско V!";    // Fixed width 8 chars
const char* SleepingSimpleString = "Сън";      // Must be <= 4 chars
const char* SleepingAdvancedString = "Хър Хър Хър..."; // <=17 chars
const char* WarningSimpleString = "ОХ!";       // Must be <= 4 chars
const char* WarningAdvancedString = "ВНИМАНИЕ! ТОПЛО!";

/*
 * #TODO change support for multibyte constants including some Unucode characters for OK and False
 const char SettingTrueChar = '✓';   // U+2713
 const char SettingFalseChar = '✗';  // U+2717
 const char SettingRightChar = 'Д';
 const char SettingLeftChar = 'Л';
 const char SettingAutoChar = 'А';
*/

const char SettingTrueChar = '√';  // ASCII 251 symbol supported? (or U+221A)
const char SettingFalseChar = 'x';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';

const enum ShortNameType SettingsShortNameType = SHORT_NAME_DOUBLE_LINE;
const char* SettingsShortNames[16][2] = {
/* (<= 11) Power source (DC or batt)          */{ "Източник", "захранване" },
/* (<=  9) Sleep temperature                  */{ "Темп.", "сън" },
/* (<=  9) Sleep timeout                      */{ "Време", "сън" },
/* (<= 11) Shutdown timeout                   */{ "Време", "изкл."},
/* (<= 13) Motion sensitivity level           */{ "Усещане", "за движение" },
/* (<= 13) Temperature in F and C             */{ "Температурна", "скала" },
/* (<= 13) Advanced idle display mode enabled */{ "Детайлен", "екран в покой" },
/* (<= 13) Display rotation mode              */{ "Ориентация", "дисплей" },
/* (<= 13) Boost enabled                      */{ "Турбо режим", "пуснат" },
/* (<=  9) Boost temperature                  */{ "Турбо", "темп." },
/* (<= 13) Automatic start mode               */{ "Автоматичен", "работен режим" },
/* (<= 13) Cooldown blink                     */{ "Мигай при", "топъл поялник" },
/* (<= 16) Temperature calibration enter menu */{ "Калибриране", "температура?" },
/* (<= 16) Settings reset command             */{ "Фабрични", "настройки?" },
/* (<= 16) Calibrate input voltage            */{ "Калибриране", "напрежение?" },
/* (<= 13) Advanced soldering screen enabled  */{ "Детайлен", "работен екран" },
};
#endif

```

Mrkvozrout, 2017-12-16:
@gkrystev Hi, current font follows unicode chars in UTF8 encoding, so there is no ASCII above 127. Also the font goes just up to U+017F (c5 precursor) so sadly no ✓✗ chars for now.

Also there is new text to translate, look for `SettingsResetWarning`.

Can you adjust the true/false chars and provide the extra text?

gkrystev, 2017-12-18:
The updated code:

```
#ifdef LANG_BG
const char* SettingsLongNames[16] =
		{
				// These are all the help text for all the settings.
				// No requirements on spacing or length.
				/* Power source (DC or batt)          */"Източник на захранване. Минимално напрежение. <DC 10V> <S 3.3V за клетка>",
				/* Sleep temperature                  */"Температура при режим \"сън\" <C>",
				/* Sleep timeout                      */"Включване в режим \"сън\" след: <Минути/Секунди>",
				/* Shutdown timeout                   */"Изключване след: <Минути>",
				/* Motion sensitivity level           */"Усещане за движение <0.Изключено 1.Слабо 9.Силно>",
				/* Temperature in F and C             */"Температурна скала <C=Целзии F=Фаренхайт>",
				/* Advanced idle display mode enabled */"Детайлна информация със ситен шрифт на екрана в режим на покой",
				/* Display rotation mode              */"Ориентация на дисплея <A. Автоматично L. Лява Ръка R. Дясна Ръка>",
				/* Boost enabled                      */"Ползвай предния бутон за \"турбо\" режим който температурата до 450C при запояване",
				/* Boost temperature                  */"Температура при \"турбо\" режим",
				/* Automatic start mode               */"Режим на поялника при включване на захранването. T=Работен, S=Режим на сън, F=Изключен",
				/* Cooldown blink                     */"След изключване от работен режим, да мига индикатора за температура докато човката на поялника все още е топла",
				/* Temperature calibration enter menu */"Калибриране на температурата",
				/* Settings reset command             */"Връщане на фабрични настройки",
				/* Calibrate input voltage            */"Калибриране на входното напрежение. Задръжте бутон за изход",
				/* Advanced soldering screen enabled  */"Детайлна информация в работен режим при запояване", };

const char* SettingsCalibrationWarning = "Уверете се, че човката на поялника е със стайна температура преди да продължите!";
const char* SettingsResetWarning = "Сигурни ли сте, че искате да върнете фабричните настройки?";
const char* UVLOWarningString = "Ниско V!";    // Fixed width 8 chars
const char* SleepingSimpleString = "Сън";      // Must be <= 4 chars
const char* SleepingAdvancedString = "Хър Хър Хър..."; // <=17 chars
const char* WarningSimpleString = "ОХ!";       // Must be <= 4 chars
const char* WarningAdvancedString = "ВНИМАНИЕ! ТОПЛО!";

const char SettingTrueChar = 'Y';
const char SettingFalseChar = 'N';
const char SettingRightChar = 'R';
const char SettingLeftChar = 'L';
const char SettingAutoChar = 'A';

const enum ShortNameType SettingsShortNameType = SHORT_NAME_DOUBLE_LINE;
const char* SettingsShortNames[16][2] = {
/* (<= 11) Power source (DC or batt)          */{ "Източник", "захранване" },
/* (<=  9) Sleep temperature                  */{ "Темп.", "сън" },
/* (<=  9) Sleep timeout                      */{ "Време", "сън" },
/* (<= 11) Shutdown timeout                   */{ "Време", "изкл."},
/* (<= 13) Motion sensitivity level           */{ "Усещане", "за движение" },
/* (<= 13) Temperature in F and C             */{ "Температурна", "скала" },
/* (<= 13) Advanced idle display mode enabled */{ "Детайлен", "екран в покой" },
/* (<= 13) Display rotation mode              */{ "Ориентация", "дисплей" },
/* (<= 13) Boost enabled                      */{ "Турбо режим", "пуснат" },
/* (<=  9) Boost temperature                  */{ "Турбо", "темп." },
/* (<= 13) Automatic start mode               */{ "Автоматичен", "работен режим" },
/* (<= 13) Cooldown blink                     */{ "Мигай при", "топъл поялник" },
/* (<= 16) Temperature calibration enter menu */{ "Калибриране", "температура?" },
/* (<= 16) Settings reset command             */{ "Фабрични", "настройки?" },
/* (<= 16) Calibrate input voltage            */{ "Калибриране", "напрежение?" },
/* (<= 13) Advanced soldering screen enabled  */{ "Детайлен", "работен екран" },
};
#endif
```